victronenergy/venus-html5-app #21: Add Solar panel part?

LeeNX, 2018-11-04:
Like to thank the people working on this fine piece of code, really lets me drop a tablet or something in our house to show off VictronEnery details without internet access.  
  
I am no developer, so I can't really help with the app, other than ask for nice features and possible give some feedback.  
  
Also hoping this is the right place to put in feature requests and bug reports? Don't remember seeing any place in the repo ReadMe, but that has changed a bit since I first looked at this month ago.  
  
Either way, this is awesome work, thanks!

mpvader, 2019-01-07:
Hey @LeeNX; my apologies for the slow reply! I was not monitoring this issue list.

Yes; we'll also add monitoring solar. Its on the list.

mpvader, 2019-01-07:
The issue to monitor is this one: #52.

Closing this one as a duplicate.

closed 2019-01-07 by mpvader

LeeNX, 2019-01-07:
@mpvader - No worries, thanks for the great work and feedback!