victronenergy/venus-html5-app #51: Extending to multiple batteries, tanks, solar, inverters and chargers

mpvader, 2018-11-27:
Needs design; and splitting up into multiple issues.

[20181127 - html5-app.pptx](https://github.com/victronenergy/venus-html5-app/files/2620410/20181127.-.html5-app.pptx)

ucorina, 2018-11-27:
Split this into the following tasks:

* [ ] https://github.com/victronenergy/venus-html5-app/issues/52 - Show solar charger info
* [ ] https://github.com/victronenergy/venus-html5-app/issues/53 - Show tank information
* [ ] https://github.com/victronenergy/venus-html5-app/issues/54 - Improve support for inverter-charger/charger/inverter information
* [ ] https://github.com/victronenergy/venus-html5-app/issues/55 - Show multiple batteries
* [ ] https://github.com/victronenergy/venus-html5-app/issues/56 - Show custom names for batteries

I think we can keep this issue for the design updates.

ucorina, 2018-11-27:
More context for the design:

## What to hide on split screens
When going to smaller screen sizes; data can be hidden and/or shown less nicely.
Data that can be hidden, in order from least important data to most important:

Batteries except for the main one
Tank levels

The other way around, the most important components are:
The main battery
AC loads
DC loads

Its OK to only show the controls (on/off & input current limit) if the screen is large enough.

## How to show 7 batteries, 3 tanks and more on one page?

For large systems, the esthetics become less important.

For large systems  like that, its (typically) a power user and/or even professional crew. That will have, besides this app, a custom SCADA type designed system.

And the role of the html5 app is to be a dependable simple mechanism to show all the data. With less focus on looking nice and simple when comparing to a simple boat with one inverter/charger and a BMV; where the html5-app is the only interface.

ucorina, 2018-12-11:
### Design proposal

[image]

### Feedback

Feedback on the proposal, after discussion today:

Active source
- Show Shore and Generator always
- Separate Solar

Inverter charger
- On / Off → Should show “Charger only” as well
  - on a boat, charger is always on, you would never turn off the inverter

General comments
- How would the new design look for a simple boat?
- It’s not necessary to show all of this in a split screen
- Send not only the changed one, but the package of the different options
  - Main - with many stuff
  - Simple boat

Custom names
- Names of the devices can be longer → theoretically up to 32 characters, but we should support in the UI up to 16 characters

Multiple batteries
- State of charge for extra batteries - where can we show that?

Tanks
  - Leave out because the goal of the system is to show the elecrtical system overview of he boat
  - Tank levels can be seen in other apps on the boat (via NMEA2000 on Garmin / Simrad)

Is it possible to have a system with no AC Loads? Yes, when only a battery is present - see System B [here](https://github.com/victronenergy/venus-html5-app/wiki/Example-system-configurations)

thomhos, 2019-01-07:
Design has been completed. Implementation is in progress and can be tracked using the above mentioned issues.

Closing this for now.

closed 2019-01-07 by thomhos