victronenergy/venus-html5-app #53: Show tank information

ucorina, 2018-11-27:
Tanks: there can be one or more tanks. Documentation is here: https://github.com/victronenergy/venus/wiki/dbus#tank-levels. Just show them in a nice way. Fluid type is available (fresh water, fuel, etcetera). Look around on internet for inspiration for icons and such. Show it as a percentage, not as litres or gallons. Ignore the status; if status is not OK; then the percentage value will be invalid anyway; so you’ll be showing dashes.

ucorina, 2018-12-05:
[image]

Listens on N/{portalId}/tank/+/DeviceInstance to get all device instances.

Then, for each, looks at:
N/${portalId}/tank/${tankId}/FluidType
N/${portalId}/tank/${tankId}/Level

mpvader, 2018-12-11:
We decided not to do it: tanks are not part of the electrical system; and the MFDs will already have a way to show tank data.

However! later when using the html5 app for offgrid systems with a diesel generator; and no MFD but just some web-browser somewhere; this might come back.

closed 2018-12-11 by mpvader

webmasterkai, 2019-08-15:
@mpvader It is possible for a MFD to show tank data if tanks are connected to the NMEA 2000 network but I don't think Venus will send out PGN 127505 (Fluid Level) if connected to a NMEA 2000 network. How does a user get tank level data on the MFD if the senders are connected to a VenusGX? I think the only way is via the console.

johanneshovda, 2020-05-22:
@mpvader i think your statement on closing this because boats have MFD's is perhaps shortsighted. 
I am using lots of Victron equipment in a bus conversion, and there is no existing MFD or any other gauges for electrical or tank sensor information. 
I would prefer to use the HTML5 interface as the main display (on a touchscreen raspberry) to show tank information in addition to the shore power/inverter/charger/battery status, especially since I have all three tanks connected to my Venus GX. 

Kind regards

webmasterkai, 2020-05-24:
The next version of Venus OS https://community.victronenergy.com/questions/49595/venus-os-v26022-available-for-testing.html includes some tank monitoring improvements including sending out 127505 (Fluid Level). The only drawback is that it sends each tank as a different virtual device but all as data instance 0. It is possible to alter the device instance but in almost all NMEA 2000 environments displays and MFDs require unique data instances. I will post back if I am able to get 126208 to successfully update the "Fluid Instance" field as mentioned in section 7.4.4 on this page https://www.victronenergy.com/live/ccgx:start#marine_mfd_integration_by_nmea_2000