victronenergy/venus-html5-app #55: Show multiple batteries

ucorina, 2018-11-27:
The strategy to use is to:
* keep showing the .system battery, as you do now
* show all batteries that you can find as .battery as well, but then make sure to take out the one that you already show as .system; except for the /1/Voltage; if its on that same service.

Why not just show all .batteries you might ask? That’s because sometimes there is not a single .battery service, in a system that has no battery monitor; but then there will still be at least some battery data available, such as voltage, under system.

How to filter the .system battery out of the .battery services? By using the path under .system that tells you which of the .battery services it uses.

How to find them: subscribe to /battery/+. And then look for /battery/[device-instance]/Dc/[channel]/Voltage, and that’s your count.

Why the /1/Voltage exception for the system battery? That’s because the .system service won’t do anything with the /1/Voltage; it only (re)publishes the /0/Voltage.

Example:
* /battery/250/Dc/0/Voltage (and /Dc/0/Current, /Dc/0/Power, /Dc/SOC and so forth)
* /battery/250/Dc/1/Voltage 
* /battery/288/Dc/0/Voltage
*  /battery/289/Dc/0/Voltage 
* /battery/289/Dc/1/Voltage

This would be 5 different battery banks. Three that only have a voltage measurement, and three with full data. Note that some paths won’t have a /0 or /1 pre-fix. For those, they belong to /0.

Things to solve on the Venus side are:
- Name for the starter battery. Right the system can only take a custom name of the  /0 battery, not of the  /1 batteries. Which we’ll fix.

thomhos, 2019-01-07:
Implemented, needs testing and code review.

ucorina, 2019-01-07:
Implementation uses new dbus path under:
`N/{portalId}/system/0/Batteries`.

closed 2019-01-08 by ucorina