victronenergy/venus-html5-app #56: Show custom names for batteries

ucorina, 2018-11-27:
* Most devices can be given a name by the installer.
* On D-Bus they are available on the /CustomName path.
Documentation is here:
https://github.com/victronenergy/venus/wiki/dbus#generic-paths
https://www.victronenergy.com/live/open_source:ccgx:d-bus#object-paths

Logic to use: if there is a Custom name: show it. If not, and there is only one battery, show Battery when its /0, and show Starter battery when its channel /1.

ucorina, 2018-12-10:
From https://www.victronenergy.com/live/open_source:ccgx:d-bus#object-paths:

> /CustomName String, containing an editable name. This can be set by a user to distinguish the device from other devices. When a product name is presented to a user, /CustomName is the preferred name to present. When /CustomName is blank, /ProductName should be presented to the user. When /CustomName is changed, it will be stored in the device if the device supports this and otherwise in LocalSettings (com.victronenergy.settings).

ucorina, 2018-12-10:
Sample values for /ProductName and /CustomName:

[image]
[image]

heeboA, 2018-12-10:
Considering the above I guess that the batteries should be identified by the device instance at least in the cases where there are multiple batteries (device instances) without custom names. E.g.

```
Battery 15v (inst 0)
Starter battery 15v (inst 0)
Battery 10v (inst n)

-> use running indices (1,2,3,4) for different instances or display device instance number
```

mpvader, 2018-12-11:
Hi; its up to the installer to name the batteries such that there is no overlap. Identifying the batteries can be done on the CCGX: no need to show instance numbers in the UI.

ucorina, 2018-12-11:
Conclusions:
- Only do the numbering if there is no custom name
- The name is now per device → the starter battery will have the same name as the main battery → this needs to be fixed in Venus OS (https://github.com/victronenergy/venus/issues/387)

ucorina, 2018-12-11:
Updated the battery names code to only show index numbers if a name is not available.
I created another ticket for supporting custom names for starter batteries, for when the feature will be built: https://github.com/victronenergy/venus-html5-app/issues/72.
Until then, I append "(Starter)" to the battery name for starter batteries.

closed 2018-12-11 by heeboA