victronenergy/venus-html5-app #63: User feedback when System DC is not enabled

ucorina, 2018-12-04:
When showing the DC Loads, we calculate the info based on data from `/Dc/System/Power`.
It is possible that this topic returns a null value if the "System DC" setting is not enabled in Remote Console > Settings > System Setup. 

We should show a message informing the user that this option needs to be enabled.

ucorina, 2018-12-10:
After our discussion about possible system configurations (see https://github.com/victronenergy/venus/issues/380) I now realize that there are situations when it's valid for "System DC" to be off. 

Can we assume in this case that we should not show the "DC loads" box? Or should it be visible but with different data?
@mpvader @martinbosma

mpvader, 2018-12-11:
Hi; yes then it can be hidden.

mpvader, 2018-12-11:
ps. for boats; it will always be enabled. And boats are for you right now the target market. But to not do double work; lets (as said a minute ago) hide it when that setting is off.

look at the setting! not at the `/Dc/System/Power` path when it comes to showing or not showing the box.

ucorina, 2018-12-11:
When we initially discussed about this ticket, the goal was to guide the user in case the system is not configured appropriately. I think this is still valid.
By hiding the "DC Loads" box, the user will still be left wondering why no data is coming in case his "Settings" are wrong.

Especially in the case where this should always be "On" for boats, I would advise we show a message to the user to enable it in the "Settings". This way, the user can fix the issue himself - whereas if we just hide the box, he will be left wondering why it is not visible.

closed 2018-12-12 by heeboA