victronenergy/venus-html5-app #80: Support for 3 phase AC systems

ucorina, 2018-12-18:
Regarding 3 phase systems: there will be three AC voltages; and three AC currents, and also three AC powers for the `AC Loads` box. And same for Generator and Shore power boxes.

[ ] How to determine when to show simple, one phase view vs all three phases? Right now we always show the sum of three phases

thomhos, 2019-01-07:
Implemented, needs testing and code review.

ucorina, 2019-01-07:
Final implementation uses:
`N/${portalId}/system/0/Ac/ConsumptionOnOutput/NumberOfPhases` to determine whether to show 1 or multiple phases for the "AC Loads" box.
`N/${portalId}/vebus/${vebusInstanceId}/Ac/NumberOfPhases` to determine whether to show 1 or multiple phases for the Shore power/Generator box.

[image]
[image]
[image]

Checked with configuration A, K and L.

closed 2019-01-08 by ucorina