victronenergy/venus-html5-app #93: Excessive number of decimals in battery SOC

osaether, 2019-01-30:
See attached screen capture. I am using Venus v2.30~23

[image]

mpvader, 2019-01-30:
Hi; yes fixed; but a big long story to explain why its not in Venus yet and I have to run.

But, you can help: if you have time pls install https://github.com/victronenergy/venus-html5-app/releases 0.8, I think the code that can be copied onto Venus is this in link:
https://github.com/victronenergy/venus-html5-app/releases/download/0.8/venus-html5-app.tar.gz

and let me know if that works better.

After 0.8 a whole rework has been done; which is still in progress and cannot be released yet; also not candidate released. More info here: https://github.com/victronenergy/venus-html5-app/wiki/Translating-system-components-into-the-HTML5-app

osaether, 2019-01-30:
@mpvader I tried 0.8 from the link you provided but the result is the same. Maybe I did not install it correctly? Here is what I did:

deleted all files under /var/www/venus/app
copied all files in the folder dist in the tar file to /var/www/venus/app
reboot

mpvader, 2019-03-05:
Hey @osaether sorry for the delay in this. It will be fixed when we release the new version / include the new version of html5 app into a new Venus OS build. I'm closing this now.

closed 2019-03-07 by tkurki