victronenergy/venus-html5-app #100: Visual Improvements

sklappal, 2019-03-18:
- [ ] Make headlines grey & smaller and show data with white, large and consistent-sized font 
- [ ] Bottom align the lower boxes when possible so that they fill a virtual rectangle
- [ ] Make background slightly darker and add a radial gradient
- [ ] Add boat image to AC Loads (and/or DC Loads) component
- [ ] Add Input Limit label underneath the limit value to make it easier to understand

[image]

[image]

pkronstrom, 2019-03-27:
All changes in [feature/visual-improvements](https://github.com/victronenergy/venus-html5-app/tree/feature/visual-improvements) branch.

Visual Changes:
* Headlines grey & smaller, data shown with larger white font
* Lower boxes fill a virtual rectangle, this looks good in most of the cases. It looks a bit weird in some of the layouts, since the rightmost column might have a large gap when there's only two components visible on the right.
* Boat image is shown in DC Loads component, both DC and AC Loads components are reordered so that they are always firstmost.
* Input limit controls have been remade according to the newer spec.
* 'Connected' symbol is removed, since whenever whenever any data is shown, the MQTT server is connected anyways (and in case of a disconnected server, a whole-page help message to enable MQTT server is shown)
* Remote Console button is hidden on narrow screens that don't support the remote console
* Darker background -- radial background was tested, but didn't work well with the low-color MFD
*  Show a shorthand name of the product, e.g. Charger: Skylla-i
* TimeToGo shown is rounded to nearest days/hours/minutes for simplicity
* Button styles tweaked a bit (smaller font, easier to reuse)
* Measurement + status text is moved to the component header for a more streamlined component (for most of the components)

And also
* Broken tests fixed
* “grey box bug” fixed
* “Layout jitter bug” fixed

**A real screenshot of the new UI**
[image]

**All components in the same view:**
[image]

pkronstrom, 2019-03-27:
In addition -- I made a crude mobile layout for mobile users. For narrow mobile screens, the scrolling is enabled and metrics component made 100% height.

[image]

pkronstrom, 2019-04-01:
These changes can be found in [release 0.10 ](https://github.com/victronenergy/venus-html5-app/releases/tag/0.10)

closed 2019-04-01 by pkronstrom