victronenergy/venus-html5-app #105: Option to disable Changing Inverter Charger modes.

Metamba, 2019-04-22:
It may be  good to have the option of disabling the changing of the Inverter Charger mode via the HTML5 app. This may prevent mistakenly changing the inverter mode.

martinbosma, 2020-04-28:
@AlexImb Can you post your suggestion, so we can discuss it here?
@mpvader What do you think about this?

AlexImb, 2020-04-28:
My suggestion is to add a lock icon at the top that disables all the buttons in the app. It can be unlocked by default, so only users that think is very likely to accidentally touch the buttons can enable it. The state can be stored in `localStorage` to be persistent for across multiple sessions on the same device. 

[image]

AlexImb, 2020-09-14:
We looked into making this more intuitive and easier to use. We considered the idea of tapping and holding to change the state of a widget but we defaulted to the lock icon. Considering the space available on some devices and the fact that the app can also be used on devices without touchscreen makes is hard to indicate that you have to hold the button. 

So the new design proposal is:

[image]

[image]

mpvader, 2020-09-14:
ok! (but make sure contrast is good enough, on the lower image contrast is really bad, and it was already not too good

AlexImb, 2020-09-14:
> ok! (but make sure contrast is good enough, on the lower image contrast is really bad, and it was already not too good

Yes, we'll also look into improving the disabled state for buttons in general as well.

AlexImb, 2020-09-30:
#153 implements the proposed solution for this issue and the improved contrast ratio for the disabled buttons:

[image]

[image]

[image]

[image]
[image]

AlexImb, 2021-03-08:
@thlassche this functionality together with the combined generator metrics + controls (https://github.com/victronenergy/venus-html5-app/issues/142) and the 3A input limit (https://github.com/victronenergy/venus-html5-app/pull/152) are on the `develop` branch and planned to be included in the Venus v2.70 release.

AlexImb, 2021-04-27:
Included in 1.7.0

closed 2021-04-27 by AlexImb