victronenergy/venus-html5-app #106: venus html5 interface broken layout and battery missing

ezraholm50, 2019-05-16:
Hey!

I've just tried to install the html5 interface, the dev testing page works, but no battery section is showing. I thought this was default. Do i need to add some code block somewhere?

Also when trying to access it on my venus device the layout is messed up. (needed to create /var/www/venus/app since ip/app hosted my remote console)

Easy Solar in home on grid config with ESS.

See screenshots.

Please advise.

[image]
[image]
[image]

LeeNX, 2019-05-16:
I notices the battery missing too, but just not had the time to ask about why it's now gone.

ezraholm50, 2019-05-24:
Trying to build this on Ubuntu:

```
wzm@desktop:~/venus-html5-app$ npm run dev

> victron-venus-metrics@1.1.0 dev /home/wzm/venus-html5-app
> webpack-dev-server --hot --inline --development -d

WARNING: We noticed you're using the `useBuiltIns` option without declaring a core-js version. Currently, we assume version 2.x when no version is passed. Since this default version will likely change in future versions of Babel, we recommend explicitly setting the core-js version you are using via the `corejs` option.

You should also be sure that the version you pass to the `corejs` option matches the version specified in your `package.json`'s `dependencies` section. If it doesn't, you need to run one of the following commands:

  npm install --save core-js@2    npm install --save core-js@3
  yarn add core-js@2              yarn add core-js@3

 ERROR  Failed to compile with 76 errors                                                                                                                                                                   9:51:48 PM

These dependencies were not found:

* core-js/modules/es6.array.from in ./src/app/components/Battery/Battery.js, ./src/app/components/Views/Metrics.js and 1 other
* core-js/modules/es6.array.sort in ./src/app/components/Battery/Battery.js
* core-js/modules/es6.object.assign in ./src/app/components/ActiveSource/ActiveInValues.js, ./src/app/components/Chargers/Charger.js and 8 others
* core-js/modules/es6.object.to-string in ./src/app/components/Fade.js, ./src/app/components/Battery/Battery.js and 10 others
* core-js/modules/es6.regexp.constructor in ./src/app/mqtt/MqttClientProvider.js, ./src/app/utils/util.js
* core-js/modules/es6.regexp.match in ./src/app/mqtt/MqttClientProvider.js
* core-js/modules/es6.regexp.replace in ./src/app/components/Views/Error/Error.js, ./src/app/mqtt/MqttClientProvider.js and 1 other
* core-js/modules/es6.regexp.search in ./src/app/components/Views/Error/Error.js
* core-js/modules/es6.regexp.split in ./src/app/components/Chargers/Charger.js, ./src/app/components/InverterCharger/InverterCharger.js and 2 others
* core-js/modules/es6.regexp.to-string in ./src/app/components/Battery/Battery.js, ./src/app/components/NumericValue/NumericValue.js and 3 others
* core-js/modules/es6.set in ./src/app/mqtt/MqttClientProvider.js
* core-js/modules/es6.symbol in ./src/app/components/Fade.js, ./src/app/components/Battery/Battery.js and 5 others
* core-js/modules/es7.array.includes in ./src/app/components/Inverters/Inverters.js, ./src/app/utils/util.js
* core-js/modules/es7.object.entries in ./src/app/mqtt/MqttClientProvider.js, ./src/app/mqtt/victron/GetPortalId.js and 1 other
* core-js/modules/es7.object.values in ./src/app/components/Battery/BatteryLevel.js, ./src/app/components/Chargers/Chargers.js and 3 others
* core-js/modules/es7.symbol.async-iterator in ./src/app/components/Battery/Battery.js, ./src/app/components/Views/Metrics.js and 3 others
* core-js/modules/web.dom.iterable in ./src/app/components/Fade.js, ./src/app/components/Battery/BatteryLevel.js and 9 others

To install them, you can run: npm install --save core-js/modules/es6.array.from core-js/modules/es6.array.sort core-js/modules/es6.object.assign core-js/modules/es6.object.to-string core-js/modules/es6.regexp.constructor core-js/modules/es6.regexp.match core-js/modules/es6.regexp.replace core-js/modules/es6.regexp.search core-js/modules/es6.regexp.split core-js/modules/es6.regexp.to-string core-js/modules/es6.set core-js/modules/es6.symbol core-js/modules/es7.array.includes core-js/modules/es7.object.entries core-js/modules/es7.object.values core-js/modules/es7.symbol.async-iterator core-js/modules/web.dom.iterable
```

ezraholm50, 2019-05-27:
Ok installing works fine now, redone it a couple of times and it now runs on my CCGX. Had to tweak some hiawatha settings.

But still the battery component is missing. Last working version was 0.6 (tried them all) But my knowledge on this topic is not enough, tried to copy and paste some sections and editing reference to files but i did not get it to work.

Any guidance will be appreciated.

ezraholm50, 2019-05-31:
Resolved upgrading to 2.3 beta

closed 2019-05-31 by ezraholm50