victronenergy/venus-html5-app #108: Raymarine home icon occludes Victron logo

martinbosma, 2019-09-13:
[image]

AlexImb, 2019-12-23:
Some possible solutions for this are:

1. Center the logo
[image]

2. Increase the vertical padding which probably does not look as good for other types of MFDs and split screens

mpvader, 2019-12-23:
Hi Alex, 👍 on all the fixes. For this one, centering that logo is quite disturbing the design, no?

martinbosma, 2019-12-23:
Agree. How about left aligning it with the left column in a similar way the remote console button is aligned on the rhs?
Don't know how that will work out in case of a single column and/or small resolution.

thlassche, 2019-12-27:
Agree with @martinbosma, @AlexImb can you make a screenshot of how that looks?

AlexImb, 2019-12-27:
W: 320
[image]

W: 500
[image]

W: 768
[image]

W: 1024
[image]

W: 2056+ 
[image]

thlassche, 2019-12-27:
Looks good to me. 

@mpvader @martinbosma do you happen to know if the Raymarine devices also has resizable widgets? I.e.: can it happen that the Home icon still occludes the Victron logo if there is a small configuration chosen on the Raymarine?

AlexImb, 2019-12-27:
From looking at their website it looks like they do and can even render more buttons on the top-left next to the home one. That's the reason I initially suggested the central logo. 

However, the current solution should reduce the cases when this happens. If there are still cases when it occurs then increasing the top padding will definitely fix it.

martinbosma, 2019-12-27:
Looking good! Afaik, the Raymarine MFDs only have full screen widgets. Perhaps in the future they will have split screens/smaller screen configurations.

Hower, the other brands do support smaller configurations. The Navico brands also can have buttons in the corners, see pic.
[image]

martinbosma, 2020-06-24:
@AlexImb I think this needs some more thinking/work. Below a picture of a 12" Raymarine MFD; the initial issue was with a much bigger screen. 

I also noticed that the top left button on a Simrad is also overlapping the Victron icon; see above for the Simrad layout. I think this applies to all Navico brands.

Perhaps we can add more top padding?

[image]

AlexImb, 2020-06-24:
Fix included in Release 1.4.2.

AlexImb, 2020-06-24:
[image]

PR #116 (2020-01-06): Align header elements with the metrics columns
Fixes: #108
closed 2020-06-24 by AlexImb