victronenergy/venus-html5-app #109: Automatically disable on/off/charger only control

mpvader, 2019-11-19:
For systems in which that feature doesnt exist (vebus bms)

martinbosma, 2019-11-19:
It should already be disabled based on `/ModeIsAdjustable`, see [here](https://github.com/victronenergy/venus-html5-app/blob/master/src/app/components/InverterCharger/InverterCharger.js#L64). Or do you want to hide it?

mpvader, 2019-11-20:
I made that issue following a complaint (from Gary) that that doesnt work. I’ll tell him that it should work and update fw etc.

mpvader, 2019-11-20:
Or perhaps the issue is indeed to hide it. Or make it more info like & less button like

AlexImb, 2020-04-28:
Current state using simulation M from [venus-docker](https://github.com/victronenergy/venus-docker):

`Message received: N/0242ac110002/vebus/257/ModeIsAdjustable - {"value": 0}`

[image]

For other values: 

[image]

marcoemanuelecocco, 2020-07-10:
in my camper the inverter is installed inside the cabin and the hum of the transformer can be heard at night. 
 It would be a great idea if you could program the times when the inverter can be turned off, only chargers, only inverters.
  A bar graph with 24 hours divided into 30-minute sections would be enough and for each section you could choose the option: off, charger, inverter

martinbosma, 2020-07-10:
> in my camper the inverter is installed inside the cabin and the hum (noise) of the transformer can be heard at night.
> It would be a great idea if you could program the times when the inverter can be turned off, only chargers, only inverters.
> A bar graph with 24 hours divided into 30-minute sections would be enough and for each section you could choose the option: off, charger, inverter

@marcoemanuelecocco Thank you for your suggestion, but I don't think this is as easy as it sounds as it also requires changes to the Venus operation system. Did you know that you can show the app on your phone/tablet as well? If you browse to `http://<ip address of the GX device>/app` or `http://venus.local/app` you can control the inverter from your bed ;).

marcoemanuelecocco, 2020-07-10:
Yes i make the control with a Raspberry and a simple LCD resistive 7"
monitor

AlexImb, 2020-09-18:
#151 solves this by hiding the buttons when `/ModeIsAdjustable` is 0 and adds the mode in the subtitle:

Before:

[image]

After: 

[image]

Another proposal is to also add a visual indicator for the mode:

[image]

[image]

For `Charger only` it can be orange. 

Alternatively, we can change the main icon of the widget if you consider that the text in the subtitle is not enough.

@martinbosma, what do you think?

martinbosma, 2020-09-18:
@AlexImb I would say the text is enough, as the colors don't add extra info. 

>Alternatively, we can change the main icon of the widget if you consider that the text in the subtitle is not enough.

How would you propose to change it? Change the color or design icons for on/off/charger only?

Just for clarification: this change is only when the mode is not adjustable, i.e. when a VE.Bus BMS is present in the system, right?
And not when the ui is locked as per #105

AlexImb, 2020-09-18:
> How would you propose to change it? Change the color or design icons for on/off/charger only?

Yes, have different icons for the modes on the same white background.

> Just for clarification: this change is only when the mode is not adjustable, i.e. when a VE.Bus BMS is present in the system, right?
> And not when the ui is locked as per #105

Yes, when the UI is locked the buttons are just disabled. This is only when `/ModeIsAdjustable` is 0.

martinbosma, 2020-09-18:
>>How would you propose to change it? Change the color or design icons for on/off/charger only?
>
> Yes, have different icons for the modes on the same white background.

I think showing the text is sufficient. I don't think changing the icon has added value as the user doesn't know he can also look at the icon.

>>Just for clarification: this change is only when the mode is not adjustable, i.e. when a VE.Bus BMS is present in the system, right?
>>And not when the ui is locked as per #105
>
>Yes, when the UI is locked the buttons are just disabled. This is only when /ModeIsAdjustable is 0.

Ok, clear. I was a bit concerned about the height of the widget changing dynamically and then the widgets are rearranging and jumping all over the place. However, `/ModeIsAdjustable` does not change unless you change the system/hardware.

AlexImb, 2020-09-18:
Ok, then we go for the text only. 

This is how it looks for simulation (m) from `venus-docker`: 

[image]

closed 2020-09-18 by AlexImb

martinbosma, 2020-09-18:
lgtm