victronenergy/venus-html5-app #110: Automaticallly test a defined list of configurations

thlassche, 2019-12-10:
Goals is to automatically test the app against a certain set of configurations, so that we can give suppliers a list of configurations that are known to be working. In order to do so:

- Set up a CI pipeline (Jenkins / Travis ?)
- Render the app on different resolutions and browsers, based on a defined list
- Gather the results (screenshots)
- Automatically determine whether the app works correctly (JS errors, rendering errors)

AlexImb, 2020-01-07:
How do we want to store the test results and artefacts such as screenshots, videos and reports? 

One option is the Cypress Dashboard that looks like this: 
https://dashboard.cypress.io/projects/gho3jq/runs

We can upload them anywhere from Travis CI: S3, other tools, etc.

martinbosma, 2020-01-08:
I'll leave this up to you. If you think Cypress works fine, we will use this. 

Regarding pricing:
Do you think the [`Seed`](https://www.cypress.io/pricing/) option works for us?

AlexImb, 2020-01-09:
The `Seed` option only has 500 runs included. We have about 80 tests per pipeline at the moment so we already used those 500 runs in about 7 pipeline runs. 

The `Sprout` version will allow us to store 25000 runs which means about 2500 pipeline runs at 100 tests per pipeline.

AlexImb, 2020-01-28:
As discussed on Slack, CI/CD was moved to GitHub Actions and the Cypress testing artifacts are uploaded for each build. 

The builds and their artifacts can be found in the Actions tab:  https://github.com/victronenergy/venus-html5-app/actions/runs/32197858

PR #126 (2020-02-03): Add CI/CD with E2E testing artifacts
Replaces Travis CI completely
Keeps the old behaviour for releases on tags. 
Closes #110
closed 2020-02-03 by thlassche