victronenergy/venus-html5-app #118: Enable MQTT instructions are outdated

martinbosma, 2020-01-07:
The screenshot and instructions shown when a MQTT connection cannot be made are outdated.

The following is shown:
[image]

But the current services page on a GX device looks like this (v2.42):
[image]

Both `MQTT on LAN (SSL)` and `MQTT on LAN (Plaintext)` must be enabled.

AlexImb, 2020-01-07:
On CCGX v2.42 in Demo mode looks like this:

[image]

I updated the instructions with that picture and rounded the borders:

[image]

I can also update the text if you think it's necessary.

martinbosma, 2020-01-07:
Where is the access level set to? If it's set to `Service` or `Superuser` then it shows `NMEA2000 on MQTT` in the Services menu. This is not available when the access level is set to `User` or `User & Installer`, which is the most common use case.

martinbosma, 2020-01-07:
And can you change the text that `Remote Control > Settings > Services > MQTT on LAN (Plaintext)` must be enabled? Instead of just `MQTT`.

AlexImb, 2020-01-07:
It was set to `Superuser` indeed. Set it to `User & Installer` and now it looks like this:

[image]

martinbosma, 2020-01-07:
Perfect!

PR #122 (2020-01-07): Replace MQTT instructions image
Closes #118
commit 2b2f83ec (2020-01-07): Replace MQTT instructions image

Closes #118
closed 2020-01-07 by thlassche