victronenergy/venus-html5-app #120: Remote console conditionally display topic path

AlexImb, 2020-01-07:
I found the following TODO in the code and I was wondering if this is already implemented so I can use the right path.

```
// TODO Add topic for SystemSetup and then change value passed to header
showRemoteConsoleSetting: `N/${portalId}/settings/0/Settings/SystemSetup/<placeholder>`
```

Perhaps one or a combination of these:
[image]

martinbosma, 2020-01-07:
You mean [this](https://github.com/victronenergy/venus-html5-app/blob/7c18c630b35efd1dc556226c44570a8fdeaa54af/src/app/components/Header/Header.js#L30), right?

I can't remember what the idea was behind this. It's to show/hide the remote console button based on the value of the property? 

Right know [it's always visible](https://github.com/victronenergy/venus-html5-app/blob/7c18c630b35efd1dc556226c44570a8fdeaa54af/src/app/components/Header/Header.js#L42), regardless of the value of the property?

martinbosma, 2020-01-07:
The correct topic is: `N/${portalId}/settings/0/Settings/System/VncLocal`

AlexImb, 2020-01-07:
Yes, that's what I meant. I'll use that topic. Thank you!

PR #123 (2020-01-07): Replace showRemoteConsole placeholder topic
Fixes #120
commit 0ab6c828 (2020-01-07): Replace showRemoteConsole placeholder topic

Fixes #120
closed 2020-01-07 by thlassche