victronenergy/venus-html5-app #125: Add Fischer Panda Generator support

martinbosma, 2020-01-13:
This issue is created to discuss the [Fischer Panda generator](https://github.com/victronenergy/venus-html5-app/pull/124) component.

@khasenklever 
Personally I think the FP component is a bit crowded. What would be the minimum that needs to be shown? Are the frequency, start voltage and the different temperatures important to show? If so, perhaps we can put them in a "submenu"?
[image]

Also, do we also want to show that the FP autostart feature is disabled, e.g. someone pressed the start/stop button on the xControl panel due to maintainance. In the Venus GUI we are [showing a toast](https://github.com/victronenergy/gui/blob/master/qml/PageAcInModelFpGenset.qml#L43) in that case.

>And any idea, why it is always directly stopping the gen again after pressing "on"?

I will look into it. It seems that you are using the correct topic.
We don't have a real genset, only the simulator we received from FP, so I'm not sure it behaves the same as a real genset.

khasenklever, 2020-01-13:
> Personally I think the FP component is a bit crowded. What would be the minimum that needs to be shown? Are the frequency, start voltage and the different temperatures important to show? If so, perhaps we can put them in a "submenu"?

You are right. In contrast to all other components it really looks a bit overfilled. I thought about putting the starter-battery in the existing meta-component "Batteries", but I don't know if that component should only list batteries that store "system-energy" (which is used by the inverter).
Grouping "extra-values" an FP-Generator provides (e.g. temperatures) into a submenu sounds good, but exceeds my poor "js-skills" at the moment.

> Also, do we also want to show that the FP autostart feature is disabled, e.g. someone pressed the start/stop button on the xControl panel due to maintainance. In the Venus GUI we are showing a toast in that case.

Totally agree to this. I ran into this "trap" several times myself.

> We don't have a real genset, only the simulator we received from FP, so I'm not sure it behaves the same as a real genset.

I encountered this behaviour initially as I uploaded the page onto the CCGX intalled in our "exhibit" for the "Boot" fair which also uses a simulator-board like yours and a Lowrance HDS 7 LIVE.
Previously I tested it with a CCGX, a Chrome-browser and a real generator-control at my workplace.
So if it's not an implementation-fault in the component it might be a problem of the simulator or it's a problem of the integrated browser-component of the MFD.

martinbosma, 2020-01-13:
>You are right. In contrast to all other components it really looks a bit overfilled.

@AlexImb Can you make a redesign proposal of this component? The main view should show the following items.
- Icon
- Title
- State
- AC info; show only L1 in case of single phase and show L1, L2 & L3 in case of 3-phase system
- On and off button

Have a look at [this](https://github.com/victronenergy/venus-html5-app/blob/master/src/app/components/ActiveSource/ActiveInValues.js) and [this](https://github.com/victronenergy/venus-html5-app/blob/master/src/app/components/AcLoads/AcLoads.js) for the layout of single/three phase.

>Grouping "extra-values" an FP-Generator provides (e.g. temperatures) into a submenu sounds good, but exceeds my poor "js-skills" at the moment.

We need think of a way to show the other info, like starter voltage, frequency, etc. Perhaps we need a second "page", like the battery component.

>I thought about putting the starter-battery in the existing meta-component "Batteries", but I don't know if that component should only list batteries that store "system-energy" (which is used by the inverter).

It is not as simple as just adding another battery to the Battery component. The list of batteries is composed elsewhere. I need to look into this if it is an option.

khasenklever, 2020-01-13:
@martinbosma forget about the button-problem. I just connected the MFD to my setup (no simulator-board) and it worked as expected.

martinbosma, 2020-01-13:
> @martinbosma forget about the button-problem. I just connected the MFD to my setup (no simulator-board) and it worked as expected.

Ok, great.

martinbosma, 2020-01-13:
> It is not as simple as just adding another battery to the Battery component. The list of batteries is composed elsewhere. I need to look into this if it is an option.

We can fairly easy add this to the battery list. @izak Can you do this? The path of the service is `com.victronenergy.genset` and the starter battery voltage you can find at `/StarterVoltage`. Can you set the default name to something like `Generator starter battery`?

AlexImb, 2020-04-29:
@martinbosma At the moment, the Fisher Panda Generator component from (#131) is separate from the [ActiveSource](https://github.com/victronenergy/venus-html5-app/blob/master/src/app/components/ActiveSource/ActiveInValues.js) component which renders the Shore Power, Grid Input and **Generator** metrics panels. Do you want to keep these components separate? My suggestion is to include the extra information and the On/Off buttons in the existing ActiveSource component which probably covers #95 as well. What do you think?

[image]

A difference between the new component and the current ActiveSource (Generator) one is the topics used: 

[ActiveSource](https://github.com/victronenergy/venus-html5-app/blob/master/src/app/components/ActiveSource/ActiveSource.js):
```
    activeInput: `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/ActiveInput`,
    phases: `N/${portalId}/vebus/${vebusInstanceId}/Ac/NumberOfPhases`,
    settings: [
      `N/${portalId}/settings/0/Settings/SystemSetup/AcInput1`,
      `N/${portalId}/settings/0/Settings/SystemSetup/AcInput2`
    ],
    current: [
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L1/I`,
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L2/I`,
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L3/I`
    ],
    voltage: [
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L1/V`,
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L2/V`,
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L3/V`
    ],
    power: [
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L1/P`,
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L2/P`,
      `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/L3/P`
    ]
```

FpGen:
```
    phases: `N/${portalId}/genset/0/NrOfPhases`,
    voltage: [
      `N/${portalId}/genset/0/Ac/L1/Voltage`,
      `N/${portalId}/genset/0/Ac/L2/Voltage`,
      `N/${portalId}/genset/0/Ac/L3/Voltage`
    ],
    current: [
      `N/${portalId}/genset/0/Ac/L1/Current`,
      `N/${portalId}/genset/0/Ac/L2/Current`,
      `N/${portalId}/genset/0/Ac/L3/Current`
    ],
    power: [
      `N/${portalId}/genset/0/Ac/L1/Power`,
      `N/${portalId}/genset/0/Ac/L2/Power`,
      `N/${portalId}/genset/0/Ac/L3/Power`
    ],
    frequency: [
      `N/${portalId}/genset/0/Ac/L1/Frequency`,
      `N/${portalId}/genset/0/Ac/L2/Frequency`,
      `N/${portalId}/genset/0/Ac/L3/Frequency`
    ],
    starterVoltage: `N/${portalId}/genset/0/StarterVoltage`,
    statusCode: `N/${portalId}/genset/0/StatusCode`,
    coolantTemp: `N/${portalId}/genset/0/Engine/CoolantTemperature`,
    windingTemp: `N/${portalId}/genset/0/Engine/WindingTemperature`,
    exhaustTemp: `N/${portalId}/genset/0/Engine/ExaustTemperature`
```

Are the values from `N/${portalId}/genset/0/Ac/*` different than `N/${portalId}/vebus/${vebusInstanceId}/Ac/ActiveIn/*` when the AcInput is a genset? 

Currently, there is no simulation including a Fisher Panda generator in [venus-docker](https://github.com/victronenergy/venus-docker) that I can use for testing and developing this. 

@khasenklever Could you please paste a link to a VRM installation that has the required devices so we can record a simulation?

martinbosma, 2020-04-29:
The values on ActiveSource component are just measurements from the Multi/Quattro. When AcInput is set as genset and this is the active input of the Multi/Quattro, these values are probably the same. However, the source is different. The `genset` values come from the FP hardware and the `vebus` values from the Multi/Quattro. 

The additional values like the coolant temperature, etc. are specific for Fischer Panda. Would you also make that part of the ActiveSource component then? 

Is the ActiveSource component also visible when it's not the active input? Because if the generator is not running/active, you still should be able to start it and see the other values, such as the temperatures.

Regarding #95 
We also have generator support using the relay on a GX device. This is a [different service](https://github.com/victronenergy/dbus_generator) than the Fischer Panda one. It is using [`generator `](https://github.com/victronenergy/gui/blob/master/dbus_service.cpp#L136), instead of [`genset`](https://github.com/victronenergy/gui/blob/master/dbus_service.cpp#L121). See also [here](https://github.com/victronenergy/gui/blob/master/qml/PageGenerator.qml). I think this service has a different way of starting/stopping than the FP one.

I'm guessing it will be quite a mess putting all this in the ActiveSource component. So perhaps it's better to have a dedicated generator control component? How do you think about that?

I was plannning to go to the office tomorrow, so I can try to make a recording with the simulation box (or take it home). Where do you need a recording of exactly? Starting, stopping, AutoStart off, AutoStart on?

AlexImb, 2020-04-29:
> I'm guessing it will be quite a mess putting all this in the ActiveSource component. So perhaps it's better to have a dedicated generator control component? How do you think about that?

Yes, I also think now that having a dedicated generator control component is better considering all the possible topics and ways of controlling generators. I'll look through the code you linked and try to make a component that can accommodate both `generator` and `genset` topics. If an installation has both (probably not a common use-case) two such components will be rendered.

> I was plannning to go to the office tomorrow, so I can try to make a recording with the simulation box (or take it home). Where do you need a recording of exactly? Starting, stopping, AutoStart off, AutoStart on?

Yes, that's enough.

martinbosma, 2020-04-29:
> Yes, I also think now that having a dedicated generator control component is better considering all the possible topics and ways of controlling generators. I'll look through the code you linked and [...]

It seems to me that this is very unlikely. So showing both is fine in that case.
Note that in case of `generator`, this service does not provide any measurement itself afaik, it only controls the generator. Measurements are visible in the ActiveSource box if this generator is connected to the Multi/Quattro.

Also, can you make the FP genset component show the AC measurements as the ActiveSource component does? It does need an extra column for the frequency, though.

And the `starterVoltage` can be removed, because it should be present in the battery box. See https://github.com/victronenergy/dbus-systemcalc-py/commit/69f32a83f2a84a594d8a125e1dc9e52ee92db40d.

>I was plannning to go to the office tomorrow, so I can try to make a recording with the simulation box (or take it home). Where do you need a recording of exactly? Starting, stopping, AutoStart off, AutoStart on?

Is it ok to make two separate recordings? One for normal operation and one where I change the AutoStart feature?

AlexImb, 2020-04-29:
> Also, can you make the FP genset component show the AC measurements as the ActiveSource component does? It does need an extra column for the frequency, though.

Yes, I'll add the frequency as an extra column. 

> Is it ok to make two separate recordings? One for normal operation and one where I change the AutoStart feature?

Yes, it's perfect.

khasenklever, 2020-05-05:
@AlexImb sorry, haven't been in the office for weeks (beeing able to set up a VRM instance for you).
Is the recording of martinbosma enough or should I ask a colleague to install a basic setup?

AlexImb, 2020-05-05:
@khasenklever the current recordings are enough for now.

AlexImb, 2020-06-01:
The current progress of the Generator control component: 

[image]

[image]

[image]

[image]

Because the https://github.com/victronenergy/venus/wiki/dbus#generator-startstop service does not seem to be exposed trough MQTT the Relay Generator uses `N/${portalId}/settings/0/Settings/Relay/Function` and `N/${portalId}/settings/0/Settings/Relay/Polarity` to get the state which I don't think is entirely correct. Also, I don't know to which topic to write to control the relay. Exposing the `com.victronenergy.generator.startstop0` will probably fix this issue. 

Now working on showing the additional values (coolant, winding, exhaust) on a separate "page" similar to the Battery component.

AlexImb, 2020-06-04:
After moving the frequency and the starter battery voltage the only extra info remaining is the temperatures. I think is not enough info to justify the addition of another page, so my suggestion is to show them like this: 

[image]

[image]

khasenklever, 2020-06-04:
Looks good to me!

AlexImb, 2020-06-08:
@martinbosma The components are implemented and look like this: 
[image]
[image]
[image]
[image]

There are is one thing that is different compared to the GUI. For the Generator Relay component when manual On/Off modes are selected the AutoStart is disabled. In the GUI this happens only for the Fischer Panda generator. I can easily change it to not disable the Auto start/stop option but that will result in 2 buttons being active at the same time.

mpvader, 2020-06-08:
Hey @both ,looking nice!

One comment: Generator Relay sounds like not the best possible (default) name

AlexImb, 2020-06-08:
Some name suggestions: 
- Generator start/stop (the same name as the GUI)
- Relay-controlled Generator
- Generator (since the AcSource was renamed from just Generator to Active Input: Generator)

mpvader, 2020-06-08:
Lets make it a simple `Generator`

martinbosma, 2020-06-08:
@AlexImb Looks nice! 

The notification about the FP AutoStart feature being disabled is now located at the bottom and seems to change the height of the component when it becomes visible. That could potentially cause components to move on the page. Isn't it better to keep the height as fixed as possible? Perhaps you can put the text at the top or show some kind of overlay message.
Also, to prevent confusion between the FP AutoStart feature and the Victron auto start/stop, perhaps you can change the text to `The Fisher Panda AutoStart functionality is currently [...]`

Tomorrow I'm at the office, and can do some brief tests.

AlexImb, 2020-06-08:
I changed the text. I don't think the height change is a problem since the position of the metrics panels is recalculated periodically and the generators are also the last metrics displayed (nothing under them). 

For testing on your GX device, I suggest to merge the `feature/fpgen` branch into `master` locally and then build and deploy on your local GX device.  I suggest this because I also merged #132 which might influence the size and positioning of the newly added components on large screens.

martinbosma, 2020-06-08:
>I don't think the height change is a problem since the position of the metrics panels is recalculated periodically and the generators are also the last metrics displayed (nothing under them).

Ok, that's fine. I've seen the layout changing when the height of some component changed. That's why I was a bit worried.

Regarding your suggestion about testing: will do

mpvader, 2020-06-11:
For any customers reading this: that this code is in master here & issue is closed does not mean its also included in a Venus OS beta yet.

We'll most likely add all this in v2.70; which takes a while: currently we're still in the v2.60 cycle.

sbuthe, 2020-07-30:
We have updated both our "Color Control GX" and our "Venus GX" to the software release "Venus OS v2.60 ~ 35". With the »Venus GX« the Fischer Panda Generator is correctly displayed on a »Lowrance HDS7 Live« and you can start and stop using the app. 

[image]

But if we use the »Color Control GX«, only a battery with the battery voltage is shown on the Lowrance panel, but no generator. Starting and stopping the generator is not possible.

[image]

Today we updated to v2.60~37, but it is all the same.

Is there a setting that we may have chosen differently with Color Control?

sbuthe, 2020-07-30:
We tested with two other MFDs. The »Raymarine Axiom 7« (software 3.12.199) shows only a blank page when the Victron app is opened. 

The "Garmin GPSMAP 722 Plus" only shows the message "Searching for OneHelm Device Victron" when the Victron app is opened. The Fischer Panda Generator is not recognized on both MFDs.

This is with both, the "Color Control GX" and the "Venus GX".

martinbosma, 2020-07-30:
@sbuthe If the Fischer Panda Generator is visible on the normal user interface, then it should also be present Lowrance display. It seems to recognize the starter voltage from the FP, but it's showing the wrong Generator widget. Did you perhaps also configure the relay of the CCGX as `Generator start/stop`? 

Perhaps you can also try to restart the Lowrance, as I've experience some caching issues when switching from one device to another.

martinbosma, 2020-07-30:
@sbuthe 
The Raymarine Axiom should be compatible; we have a Axiom 12 here and it worked perfectly for a long time. Victron support should work since Lighthouse v3.11. Please check if the IP settings of the GX device are set to automatic.

According to Garmin, the GPSMAP 722 Plus should be compatible with Victron, but I cannot test it here as we only have a GPSMAP A12. Please also verify the IP settings. Note that the Garmin has a build-in DHCP server, which cannot be disabled.

sbuthe, 2020-07-30:
@martinbosma Thanks for the information. On the Raymarine Axiom the Victron App appeared after changing the IP settings of the GX devices to automatic. 

The problem with the CCGX still exists. Althoug the relay of our CCGX is not configured as Generator start/stop, the Victron App Shows the wrong widged. This is on both MFDs, Lowrance and Raymarine.

martinbosma, 2020-07-30:
>On the Raymarine Axiom the Victron App appeared after changing the IP settings of the GX devices to automatic.

Ok, so the Raymarine, Lowrance and Garmin all show the Victron app now? No more white screens or `Searching for OneHelm Device Victron` message?

So, the only issue remaining is that the wrong widgets are shown? We're looking into this already.

sbuthe, 2020-07-31:
The Victron App is discovered on Lowrance and Raymarine. 

With the Garmin we tested the following:
1.	The VenusGX is switched of.
2.	Starting the Victron-App on the Garmin device – “Searching for OneHelm Device Victron” is displayed.
3.	Starting the VenusGX (supply voltage on).
4.	The Garmin shows the Victron App with Fischer Panda Generator data.

We repeated this procedure with the CCGX and the Garmin behaved in the same way, with the restriction that only a battery is displayed, as the other panels do. It seems the Garmin needs the VenusGX / CCGX start up?