victronenergy/venus-html5-app #128: For certain aspect ratios the MQTT instructions are out of the viewport

AlexImb, 2020-02-04:
[image]

A solution for this will be to show two columns (instructions on the left and VNC image on the right) for wide aspect ratios with a low height.

martinbosma, 2020-02-05:
Can you make a mockup of the two column option to see how it looks? 

And what happens with screens with a "normal" aspect ratio, but low resolution? Is the text and image then simply scaled down?

AlexImb, 2020-02-17:
> And what happens with screens with a "normal" aspect ratio, but low resolution? Is the text and image then simply scaled down?

It was not scaling down. Not it does. Combined with the 2-column view for low height it looks like this:

[image]

[image]

[image]

commit 5833da46 (2020-02-17): Improve responsiveness of mqtt-instructions

Fixes #128
closed 2020-02-17 by thlassche