victronenergy/venus-html5-app #135: Show Remote Console button when showing MQTT instructions

martinbosma, 2020-06-11:
Since the `Remote Console` does not depend on MQTT, we can show the button when the app is not working and the MQTT instructions are shown. This makes it a lot easier for the customer to enable the required settings compared to for example using the GX Wi-Fi access point.

Perhaps it can also made clear in the text that the user can press the `Remote Console` button on this page.

[image]

AlexImb, 2020-06-17:
In https://github.com/victronenergy/venus-html5-app/issues/120 we made the Remote console button dependent on the `/Settings/System/VncLocal` topic. 

Now, in https://github.com/victronenergy/venus-html5-app/pull/137, I added the Remote console button (default header) when the MQTT instructions are displayed, independent of the `/VncLocal` value. 

It looks like this: 

[image]

When the Remote on LAN option is disable it looks like this: 

[image]

After the MQTT connection is established and the `/VncLocal` value becomes available the Remote console button is hidden:

[image]

@martinbosma, what do you think? Also, should I mention in the instructions that the Remote Console has to be enabled on LAN or let the users discover that after clicking on the Remote Console button?

martinbosma, 2020-06-17:
Hmm, I forgot about #120. 

Nevertheless, I think this is the best we can do. In most cases the `Remote Console on LAN` will be enabled, especially when using headless GX devices, like the VenusGX or CerboGX, where it's enabled by default.

The goal of this change is to provide a convenient way to enable MQTT, so the user doesn't have to take his laptop or phone and connect the local WiFi AP of the GX and browse the it's IP address. If `Remote Console on LAN` turns out to be disable, we should point them in the right direction.

What do you think about changing the text to:
>Could not connect to MQTT server.
>Please check that MQTT is enabled in your settings: 
>_Settings > Services > MQTT on LAN (plaintext)_
>Try using the _Remote Console_ button at the top of this screen to change your settings remotely.
>Make sure _Settings > Remote Console > Enable on LAN_ is enabled.

Or perhaps you can reword it a bit better.

AlexImb, 2020-06-17:
> > Could not connect to MQTT server.
> > Please check that MQTT is enabled in your settings:
> > _Settings > Services > MQTT on LAN (plaintext)_
> > Try using the _Remote Console_ button at the top of this screen to change your settings remotely.
> > Make sure _Settings > Remote Console > Enable on LAN_ is enabled.

This might be interpreted that the Remote Console > Enable on LAN setting is required for the html5-app to work. Considering that the setting is likely to be enabled and that the Remote Console screen says "(...) Make sure to enable console on LAN", I'll omit it from the MQTT instructions screen.

PR #137 (2020-06-17): Add remote console button without MQTT
Closes #135 when merged.
closed 2020-06-17 by AlexImb