victronenergy/venus-html5-app #136: Not an issue, just a thank you!

KidA001, 2020-06-12:
I want to say thank you to the Victron devs. I have been continually impressed by all the products Victron puts out. As a developer I was drawn to Victron as you were one of the few (maybe only) solar/off-grid companies that made decent software to pair with their hardware. That excitement only grew when I found out about VenusOS on RaspberryPi, node-red...etc. 

We're completely off-grid, so I use the venus-html5-app on a touchscreen in our home to monitor our power system. I made some modifications to the react application and I was thoroughly impressed with your code. Clean, modular and easy to work with. Exceptional work.

Thank you for open sourcing your software. It's made me a victron customer for life and puts you way. ahead of everyone else in the field. 

I wasn't sure of any other way to reach out to the devs on this project, so hopefully this is ok, I'll close this "issue" in a couple days or feel free to do so yourself. Thank you!!!

mpvader, 2020-06-12:
Haha well appreciated @KidA001 ! I’ll make sure all involved developers receive this compliment.

This a good way to start the day!

Thank you, Matthijs

KidA001, 2020-06-12:
@mpvader thanks for passing it on!

closed 2020-06-12 by KidA001