victronenergy/venus-html5-app #138: Add paginated views

AlexImb, 2020-06-23:
The current implementation hides the metrics that are not inside in the viewport. Due to the increasing number of metrics/widgets this might be a problem for some complex installations. 

A solution for this will be to add pages and controls (and gestures) to navigate trough them.

@martinbosma What do you think?

AlexImb, 2020-06-29:
[image]

This is how it looks until now. Still to implement:

- [ ] swipe gestures for switching between the pages

mpvader, 2020-06-29:
Hi, I haven't checked what data then gets pushed to a second page - but take care that a lot of effort went into the original design to get all information one page.

johanneshovda, 2020-06-29:
You could consider pagination within each component/section, instead of the entire page. 
That way, you can see e.g the battery section, which would then paginate to show more than X number of batteries. 

I would suggest the importance of seeing all components on one page, rather than not be aware of status on a secondary page you might not realize existed..

AlexImb, 2020-07-01:
The pagination is active only when the components don't fit on the screen after reducing their height trough other means such as reducing the margins and paginating the battery list. The screens that already show all the metrics on one page won't be affected. The current implementation, as well as the previous ones, hide the components without any way of seeing or knowing that they are being hidden. 

On small and split screens only the first 2-3 components are currently displayed:

[image]

By adding pagination, all the components can be displayed on those small screens as well: 

[image]

[image]

The pagination of individual components as used in the Batteries component is not really applicable for the other components. Maybe the ones that are displaying information for all 3 phases can make use of that:

[image]

Having paginated screens and arrows to navigate trough them seems to be the pattern also used by the native apps running on the MFDs: 

[image]

@mpvader @johanneshovda @martinbosma What do you think? Should I continue with finishing the screens pagination or look into alternative solutions?

martinbosma, 2020-07-02:
@AlexImb I think this already looks is nice and is much better then just hiding widgets without any indication. I'm fine with proceeding working on this solution.

What would be the alternative solution(s) then?

AlexImb, 2020-07-02:
> What would be the alternative solution(s) then?

To find a way to fit all the information on the screen at any size, which might involve hiding icons, reducing margins/padding, etc. However, I think that pagination is a better solution long-term because new metric components will probably be added later on.

martinbosma, 2020-07-02:
I agree

PR #140 (2020-07-16): WIP: Screens pagination
Implements and closes #138
closed 2020-07-16 by AlexImb