victronenergy/venus-html5-app #141: App update in Venus GX hardware

marcoemanuelecocco, 2020-07-10:
in the victron venus GX how can I update the app, since I only find a www / app folder with a very small part of the interface?

martinbosma, 2020-07-10:
Hi @marcoemanuelecocco, the HTML5 app is shipped with VenusOS and will be automatically updated once we decide to include a new version. The next update for the HTML5 app is scheduled for Venus v2.60.

If you manually want to update the app, I would recommend to follow [these ](https://github.com/victronenergy/venus-html5-app/blob/master/README.md)instructions. Note that a manually uploaded version will be overwritten with the version from Venus in case of a firmware update.

closed 2020-07-10 by martinbosma

marcoemanuelecocco, 2020-07-10:
I've installed the 2.6.30 but don't see the update about pagination...

martinbosma, 2020-07-10:
The screen pagination is not included in Venus yet, at it is still work in progress.