victronenergy/venus-html5-app #142: Combine the Generator control component with the ActiveSource

AlexImb, 2020-07-28:
At the moment, the component that controls the generator using the internal relay of a GX device (Generator) is separate from the one that displays the measurements from the connected AC Inputs (ActiveSource).

[image]

AlexImb, 2020-07-30:
The combined Generator control and AC Input (ActiveSource) is implemented and has the following states:

AC Input 1: Generator
Relay Function: not Generator Start/stop

[image]

AC Input 1: Generator
Relay Function: Generator Start/stop

[image]

AC Input 1: Not available
AC Input 2:  Not available
Relay Function: Generator Start/stop

[image]

AC Input 1: Generator
AC Input 2: Generator
Relay Function: Generator Start/stop

Note: The buttons from both metrics control the same Generator (`/0/Generator0`) because it is not possible to know which generator is connected to which AC Input.

[image]

With data:

Simulation L: Quattro + BMV - boat with generator - three phase

[image]

Simulation F: Quattro + BMV - boat with generator - single phase

[image]

The Fisher Panda Genset is still a separate component so the following case in which another Generator is connected to the AC Input is theoretically still possible: 

[image]

If the FP Genset will be the one connected to the AC Input then the measurements will be displayed in both components. However, they will come from different sources: `/genset/0/Ac/*` for the FP Genset and `/vebus/${vebusInstanceId}/Ac/*` for the Generator one. 

Hiding Generator component(s) when the FP Genset component is present is an option, but considering that it is not possible to know to which AC Input is connected or if it is connected at all (as far as I know) I would suggest to keep both. 

@martinbosma What do you think?

martinbosma, 2020-07-30:
@AlexImb Looks good!

>Hiding Generator component(s) when the FP Genset component is present is an option, but considering that it is not possible to know to which AC Input is connected or if it is connected at all (as far as I know) I would suggest to keep both.

Yes, keep both. In theory there can be loads connected directly to the generator, which will not be visible on the AcSource widget.

AlexImb, 2020-08-10:
Closed via #143

PR #143 (2020-08-10): Add combined generator control and metrics component
Closes #142 when merged.
closed 2020-08-10 by AlexImb