victronenergy/venus-html5-app #149: AC Loads/In: Hide 3rd phase in case of 2-phase system

martinbosma, 2020-08-27:
The `AcLoads` & `ActiveSource` widgets show invalid values for the 3rd phase in case of a 2-phase system.

**Fix**
Show as many phases as reported by `N/${portalId}/system/0/Ac/Consumption/NumberOfPhases`
See [AcLoads](https://github.com/victronenergy/venus-html5-app/blob/master/src/app/components/AcLoads/AcLoads.js#L35) & [ActiveSource](https://github.com/victronenergy/venus-html5-app/blob/master/src/app/components/ActiveSource/ActiveSource.js#L53).

[image]

See [original post](https://community.victronenergy.com/questions/60074/venus-os-v26041-available-for-testing.html?childToView=60070#answer-60070) on community.

AlexImb, 2020-09-16:
@martinbosma I made a fix for this in #150. 

ActiveSource and GeneratorRelay use: `N/${portalId}/vebus/${vebusInstanceId}/Ac/NumberOfPhases`
AcLoads uses: `N/${portalId}/system/0/Ac/Consumption/NumberOfPhases`
GeneratorFp uses: `N/${portalId}/genset/0/NrOfPhases`

martinbosma, 2020-09-16:
Ok, unfortunately we don't have a 2-phase system to test on. Perhaps we (or you) can make a simulation in the docker image.
Or perhaps you have some test scripts and were already able to test this.

AlexImb, 2020-09-21:
I added some unit tests that simulate 1, 2, and 3 phase systems for AcLoads. Looks like this:

[image]

[image]

[image]

closed 2020-09-21 by AlexImb