victronenergy/venus-html5-app #155: Correct way to find device id

KidA001, 2020-11-25:
I modified `Solar.js` to show the state-of-charge as I often wanted to know what state my charge controller is in. What I don't love is the hardcoded device ID for my Solar Charge Controller `N/${portalId}/solarcharger/289/State`. Is there a way for me to programmatically identify or see a list of available solar charge controllers I can subscribe to?

Sorry if this isn't the place for the question. I [posted](https://community.victronenergy.com/questions/68456/programmatically-select-solarcharger-device-id.html) on the Victron Community forums under modifications but didn't receive any answers.

Modified code is below:

```typescript
const solarChargerStateDisplayName = {
  0: 'Off',
  2: 'Fault',
  3: 'Bulk',
  4: 'Absorption',
  5: 'Float',
  6: 'Storage',
  7: 'Equalize',
}

const getTopics = portalId => {
  return {
    power: `N/${portalId}/system/0/Dc/Pv/Power`,
    current: `N/${portalId}/system/0/Dc/Pv/Current`,
    solarChargerState: `N/${portalId}/solarcharger/289/State`
  }
}

const Solar = ({ current, power, solarChargerState }) => {
  return (
    <HeaderView icon={require("../../../images/icons/icon_solar.svg")} title="Solar">
      <MetricValues>
        <NumericValue value={current} unit="A" precision={1} />
        <NumericValue value={power} unit="W" />
        <span class='value'>{solarChargerStateDisplayName[solarChargerState]}</span>
      </MetricValues>
    </HeaderView>
  )
}
```

(p.s. I really love working on this application, you all have done a great job coding it. I've made some slight modifications to the dashboard so that I can display my Hot Tub temperature and turn it on and off :) )
[image]

thlassche, 2020-11-25:
Without checking it thoroughly I'd say if you have only one solar controller you can swap the instance number with a path level wildcard, a '+'-sign, so: `N/${portalId}/solarcharger/+/State`

mpvader, 2020-11-25:
And FYI, there is a fix planned for Venus OS to make those IDs static. Basically, Venus OS will reserve an ID for a certain MPPT serial number, and re-use that.

Instead of the method used now that can (only sometimes) cause renumberings

thlassche, 2020-11-25:
And some more context on how we handle it elsewhere in the app, see this source code: src/app/mqtt/victron/GetInverterChargerDeviceInstance.js

KidA001, 2020-11-25:
@thlassche @mpvader thanks for the feedback. I tried the MQTT `+` wildcard before and it throws an error, but I think what you're doing in `GetInverterChargerDeviceInstance.js` might help. I'll see if I can replicate another object to do something similar for my charge controller.

@mpvader I would love if Venus moves to static/UUID's for devices. I've had my device-id chance once before on me which is why I wanted something more reliable. Thanks

closed 2020-11-29 by KidA001