victronenergy/venus-html5-app #178: AC charger show wrong fault state

martinbosma, 2021-09-28:
This (non-VE.Bus) charger shows `VE.Bus fault condition` state, while this should just be `Fault`.
[image]

martinbosma, 2021-09-28:
It seems that same decription was used for VE.Bus and non-VE.Bus products:
https://github.com/victronenergy/venus-html5-app/blob/1ac805811241cce6e69e553f0c85d3b155c8cf6e/src/app/utils/util.js#L57
https://github.com/victronenergy/venus-html5-app/blob/012b0309d0993482d7e7ebece279050962cdce3e/src/app/locales/languages/nl.json#L59

Perhaps this can simply be refactored to `"veBusFault": "Fault",` or `"fault": "Fault", `

ni3k, 2021-09-28:
implemented the suggestion

closed 2021-09-28 by ni3k