victronenergy/venus-html5-app #192: Remote console small on high resolution displays

martinbosma, 2022-05-04:
Got a complaint from a customer and also confirmed on a Garmin 8412 that remote console is quite small.
The regular widgets do scale fine, but for the remote console there is some room for improvement.

Screenshot of Garmin 8412; it's a bit hard to see the actual size of the remote console here.
[image]

Below a few screenshots to compare different screen sizes. 
The absolute size obviously also depends on the physical size of the screen, so remote console will be much smaller on a 12" screen (Garmin 8412) than on a 24" screen (Garmin 8424) with the same resolution.

Screenshot 1920x1280, similar to Garmin 8412 (12")
[image]

Screenshot 1280x800, similar to Garmin A12 (12")
[image]

JoshuaDodds, 2022-07-24:
Hi there, 

Could you checkout this fork and see if it fixes the issue for you?

https://github.com/victronenergy/venus-html5-app/pull/200

Let me know if i can help with anything!

octafishmedia, 2022-09-14:
I have this issue on a 9" Garmin screen. Is there a solution released? A "200%" or "max-width" button would be great.

martinbosma, 2024-02-22:
This was fixed in v3.x.x somewhere

closed 2024-02-22 by martinbosma

octafishmedia, 2024-02-22:
It was indeed - v. 2.30! Now we’re talking.