victronenergy/venus-html5-app #286: Device tile has no buttons

Saharel001, 2023-04-13:
My Multiplus device has no buttons yet. I use the Marine2 beta.
[image]

if i switch to in .env.local to Marine it looks fin
[image]

martinbosma, 2023-04-13:
Hi @Saharel001 the new version is still work in progress. When we release the first version of the new design we will make sure there will be working buttons.

closed 2023-04-13 by martinbosma

Saharel001, 2023-04-22:
Reopen: First release an the buttons still missing.

martinbosma, 2023-04-22:
Yes, thank you for reporting. We're aware of it. Although this is now a release, it has not yet been fully tested and no final release has been made. We needed a release build to be able include it (temporarily) in a public beta Venus OS v3.00. This way people can see it, test it and give feedback.

Keep an eye on the Venus OS beta release thread on community. You can then give feedback there instead of creating issues here.

The final release of Venus OS v3.00 will not include the new version of html5 app, as it far from final.