victronenergy/venus-html5-app #290: Problem with Manifest on iOS

freakent, 2023-04-30:
In Safari on iOS you can open a WPA then install it by doing a "Add to Home Screen". Your manifest file is server from <host>/app/Marine/ so iOS assumes this is the path to the app. iOS does not let you change this. When you tap the icon the app gives a 404 error as it can not find a resource at /app/Marine. The spec says manifest files should be served from the root of your application, which would be /app. You should change the path to the manifest inside the html head or update routing so the home page is also available at /app/Marine.

AlexImb, 2023-08-14:
Hi @freakent,

We are going to check this. 

Thank you for reporting.

frankwaalkens, 2023-11-08:
Implemented in Release 3.1.2-beta Marine2

commit 6143d0b8 (2024-09-27): fix: Make the PWA work correctly, closes #290
closed 2024-09-27 by mman