victronenergy/venus-html5-app #321: Change the connection monitoring implementation to use the new heartbeat topic.

AlexImb, 2023-09-04:
The MQTT keepalive mechanism changes:

`N/<vrmportalid>/system/0/serial`: retained and tells you the portal id. Needed as retained to get going since portal id is part of all topics paths.

`R/<vrmportalid>/keepalive`: write there to keep the thing going. Not retained. This used to be retained, cq. is being sent retained by dbus-mqtt, to allow a client to see that the systems supports selective keep alives - which is a feature we had for a while in dbus-mqtt; but that feature of selective keepalives is now dropped.

`N/<vrmportalid>/heartbeat`: listen to this to see if the connection is up. GX devices sends a timestamp in there every 3 seconds as long as its being kept alive by something.

mman, 2024-09-19:
@AlexImb I believe this bug is already addressed by moving to FlashMQ. Is it still valid?

mman, 2024-09-19:
I have experimented with the `/heartbeat` path and I can easily observe it and or visualise in the interface. The question is whether there is a scenario where the MQTT connection is still alive (on the TCP level) but no updates are being sent. If it is a valid scenario to be concerned about.