victronenergy/venus-html5-app #322: As a user, I don't want to see the input current limit in my device list

AlexImb, 2023-09-04:
[image]

Status: Ongoing discussion. It's implemented as per design but might need to change

Internal discussion: https://victrondevelopment.slack.com/archives/CDBK9A71S/p1682142496123079

martinbosma, 2023-09-18:
Another report from Community:
https://community.victronenergy.com/questions/231599/venus-os-v3202-available-for-testing.html

>Under Devices, not sure what the line "Bulk charging" is. If this is the Multi's charger output (to battery, it should be labeled as such. But this is not consistent with what's happening in my system at the time of the screen shot as the Multi is charging the battery at about 900 watts.
[image]

mpvader, 2023-09-20:
Ok

closed 2023-10-11 by frankwaalkens