victronenergy/venus-html5-app #329: Landscape on tablet needs to be improved

martinbosma, 2023-09-18:
The tablet doesn't show the whole app in landscape mode, which it should. Like on the MFD's.
In portrait mode it is better.

Screenshots will follow.

Suggestion:
Perhaps add a button to go full screen, but only for mobile devices. It should not be shown on the MFD's or on a desktop browser.
Optionally the full screen setting could be stored.

mman, 2024-09-19:
@martinbosma What resolution that tablet has, can you still reproduce? The landscape layout should be cut in half vertically and this is how it works when I test it in the browser... so should be fixed.

martinbosma, 2024-09-20:
I tracked down the original email but no mentioning of a resolution nor brand. I've asked again.

mpvader, 2024-09-21:
Hey both, its this one: https://sailproof.shop/product/refurbished-sp10as-10-inch-android-rugged-tablet/

mpvader, 2024-09-21:
And my tablet indeed

mman, 2024-09-27:
@mpvader Could you please test one of the recent versions? I have been testing it on my devices and when in landscape mode the screen is never cut/split horizontally... always vertically as you would expect. I think this may have been already fixed.