victronenergy/venus-html5-app #331: Generator start/stop does not work as expected

martinbosma, 2023-09-18:
There are several issue with the generator start/stop:
- When the generator is started in the HTML5 app it takes it out of the auto-start mode. Which is wrong; you should be able to start it manually without taking it out of auto start/stop.
- Then in the HTML5 app it shows AC voltages at the bottom… where does it get those / ie what is the logic for that and is that properly documented in our Venus UI doc as well as implemented as such? There is no AC input configured as being a generator.
- Lastly, when the generator is stopped from the html5 app, the app sets it back into auto-start, but forgets to stop actually disable the manual start option.

HTML5 app
[image]

This also needs some UX/UI design improvements.

martinbosma, 2023-09-18:
Another report from Community:
https://community.victronenergy.com/questions/231599/venus-os-v3202-available-for-testing.html

>The Generator detail has a single button (Auto start/stop) which leads to a button set On/Off/Auto start/stop. Can't these be moved to the devices page? It's also not clear how to manually start the generator because of the button label.

mpvader, 2023-09-19:
Carefule, there is a design update coming for this. For guiv2 it already exists:

In the new design, first you see this:

[image]

And then this when pressing manual start:

[image]