victronenergy/venus-html5-app #336: I want to see Watts instead of Amps

martinbosma, 2023-09-18:
From Community:
https://community.victronenergy.com/questions/231599/venus-os-v3202-available-for-testing.html

>
[image]
I can't find a way to select power as apposed to current. I realize that many want to see current but power is more constant across multiple subsystems (AC in/out and battery or PV charger.

During the design state this topic has also been addressed, but no action taken yet.
We could provide an option to toggle between amps and watts in the views.

mpvader, 2023-09-19:
pls check with @jepefe  and Serj on how we're handling this in guiv2.

mr-manuel, 2024-02-24:
Any updates on this? I would also like to see watts instead of amps.

martinbosma, 2024-02-26:
Hi @mr-manuel, no update I'm afraid. We are working towards a performance upgrade, including some minor bug fixes and improvements. This is not part of it.

martinbosma, 2024-09-20:
I'd opt to rely on the same setting as gui-v2 as mr-manuel is suggesting, which is `/Settings/Gui/ElectricalPowerIndicator`, where 0=Watts and 1=Amps. However, in gui-v2 that setting seems to be broken, as it is not accepting Amps as value in the settings menu.

mman, 2024-09-20:
I'll try to infer the behaviour from settings... that will make things super simple :)

mman, 2024-09-27:
Step 1. complete here: https://github.com/victronenergy/victron-mfd-modules/commit/f140d0bfee60430fc3f523d501a2ce8ea219e866

mman, 2024-09-27:
Step 2. available via https://github.com/victronenergy/venus-html5-app/pull/437

[image]

[image]

closed 2024-09-27 by mman