victronenergy/venus-html5-app #337: Device name not always shown in devices component on the System Overview

martinbosma, 2023-09-18:
From Community:
https://community.victronenergy.com/questions/231599/venus-os-v3202-available-for-testing.html

>Under Devices, not sure what the line "Bulk charging" is. If this is the Multi's charger output (to battery, it should be labeled as such. But this is not consistent with what's happening in my system at the time of the screen shot as the Multi is charging the battery at about 900 watts.
[image]

If there is a custom name set, it should show this. Otherwise it should show the product name, I.e. `MultiPlus ....`.

martinbosma, 2023-09-18:
For above system, the custom name is indeed empty:
[image]

It should check for an empty custom name on the `CustomName` topic and show it if not empty, otherwise show the product name on the `ProductName` topic. Do note that can be a very long name, so it should be truncated in some/most cases.

martinbosma, 2023-10-09:
This seems to be some kind of race condition, which was exposed by using flashmq instead of mosquitto. Using mosquitto the data comes in very slowly when the keepalive is sent. With flashmq, however, all data is received almost instantly. Not only the vebus `ProductName` and `State` items are affected, but likely more. I've seen the shore power data not being populated, while the data was available on mqtt.

frankwaalkens, 2023-10-12:
I found the bug for the missing deviceName. The bug is located in the mfd-modules and prevents updates for the latest data for inverter-chargers.

I made a fix thats coming next week (almost done). I have to double check if this also fixes stale state data (bulk charging) and incorrect shore power data.

frankwaalkens, 2023-10-16:
Scope changed slightly, I will also compare the app with mosquitto vs flashmq to see differences in the UI. I will discuss these differences with @martinbosma and fix them in this ticket.

closed 2023-10-19 by frankwaalkens