victronenergy/venus-html5-app #338: Wrong inverter/charger data shown if there are multiple VE.Bus systems

martinbosma, 2023-09-18:
If there are multiple VE.Bus systems present, the Marine app uses data from the wrong system.
This applies to the AC input, AC load and device controls (mode and input current limit).

The logic to select the correct VE.Bus system should not be [determined by the Marine app itself](https://github.com/victronenergy/victron-mfd-modules/blob/master/src/Modules/Vebus/Vebus.provider.ts#L42), but should be taken from the system service (`system/0/VebusInstance`) instead. That is the only way to assure that the VenusOS normal interface, VRM and the Marine app are using the same VE.Bus instance.

mman, 2024-09-19:
@martinbosma I have been in VE.Bus selection code recently and can fix this if you give me a test system to validate the fix.

martinbosma, 2024-09-20:
Let me see if I can find a test system of set one up for you.

martinbosma, 2024-12-24:
@mman I've [updated](https://github.com/victronenergy/venus-docker/commit/e8bbaf1a8598c7156efe6e3f7faf19c8723977b7) venus-docker with an option to specify an additional vebus interface.