victronenergy/venus-html5-app #341: Add support for multiple VE.Bus systems

martinbosma, 2023-09-18:
Currently only one VE.Bus system is supported. 
When there are multiple VE.Bus systems, it seems the Marine app is locking onto the wrong device (see #338). 
We should of course support multiple, but this does need some thinking how to present the data.

Affected components:
- Energy - AC input
- Energy - AC loads
- Devices - Generator (when the AC input of an inverter/charger is configured as generator)
- Devices - Inverter/Charger (mode + input current limit)

martinbosma, 2024-12-24:
@mman I've [added](https://github.com/victronenergy/venus-docker/commit/e8bbaf1a8598c7156efe6e3f7faf19c8723977b7) an option to venus-docker to specify and additional vebus interface.

You can iterate over all the `vebus/#` services, where the main one is defined in `system/0/VebusInstance`. Recently also `system/0/Devices/NumberOfVebusDevices` was added, which might be helpful.

Wrt to design: in all our other user interfaces we only display the main vebus system. So, my proposal would be the same for the html5-app: show the main one in the energy and device component and the additional vebus systems only in the devices component. Here we just show the controls and the state. If needed, we can add AC data later.

mman, 2024-12-24:
Thanks, sounds like a plan.

mman, 2025-01-17:
Fixed by detecting main VE.Bus device (to show in Energy Overview) from `system/0/VebusInstance` and allowing all VE.Bus inverters and inverters/chargers to be displayed in the Devices Overview, along with their appropriate AC input current limit options. Mode setting also works for each and every VE.Bus device appropriately.

One remaining issue is that whenever there is at least one configured AC input found, it is displayed as "Shore Power" regardless of whether it is set to GRID, GENERATOR, SHORE, ... see https://github.com/victronenergy/venus-html5-app/issues/466

closed 2025-01-17 by mman