victronenergy/venus-html5-app #357: temperature sensors not shown the environmental overview

martinbosma, 2024-02-14:
Temperature only devices don't seem to be shown in the environmental overview.

Present on mqtt:
[image]

Also on gui-v1:
[image]

But not on the marine app...

However, Ruuvi tags are shown, but there are of the same service type as the temperature sensor, i.e. temperature.
[image]
[image]

mr-manuel, 2024-02-19:
Hello, I have the same issue and found out, that the Environment tile is only shown if the pressure is not `null`. Any number is accepted, even negative ones. Since only the Ruuvi tags have a pressure and they are the most used sensors this was not very obvious to find.

I checked the documentation and the pressure value should not be needed: https://github.com/victronenergy/venus/wiki/dbus#temperatures

You can make a simple test and send the payload `{"value": null}` to `N/{VRMID}/temperature/28/Pressure` and the tile should disappear. I made only the reversed test, since my temperature sensors were not shown. I send the payload `{"value": 0}` and the tile instantly appeared.

martinbosma, 2024-02-19:
Hi @mr-manuel I can confirm the same with the `N/{VRMID}/temperature/*/Humidity` topic. We'll look into this.

mr-manuel, 2024-02-19:
I searched a bit and probably the issue is in this script: https://github.com/victronenergy/venus-html5-app/blob/master/src/app/Marine2/components/boxes/EnvironmentOverview/EnvironmentOverview.tsx

fwaalkens, 2024-02-20:
Resolved in: https://github.com/victronenergy/venus-html5-app/pull/360. @mr-manuel thanks for your research efforts, should be fixed now.

closed 2024-02-20 by fwaalkens

mr-manuel, 2024-02-20:
Thanks, Is there a way to download a alpha/beta version that is already compiled? Would be nice to have the possibility as for gui-v2. It can be automated with a GitHub action.

martinbosma, 2024-02-20:
I'm afraid we don't have it available for every change, only for a release. The best we can do, is to make this part of VenusOS v3.30~xx asap.

mr-manuel, 2024-02-20:
@mpvader would it be an idea to also add a GitHub workflow to this repo that builds a new version once a day, week or any other action? Maybe this is not wanted.