victronenergy/venus-html5-app #368: Show system battery as first battery

martinbosma, 2024-05-03:
The system battery should be shown as first battery. 
The battery data is published on `system/0/Batteries` as a json object.
The system battery can be recognized by the `active_battery_service` property set to `true`.

```json

{
  "value": [
    {
      "active_battery_service": false,
      "current": 0,
      "id": "com.victronenergy.battery.socketcan_can0_vi3_uc314664",
      "instance": 3,
      "name": "Lynx Smart BMS Bottom",
      "power": 0,
      "soc": 99.98999786376953,
      "state": 0,
      "voltage": 13.5
    },
    {
      "active_battery_service": false,
      "current": 0,
      "id": "com.victronenergy.battery.socketcan_can0_vi2_uc314657",
      "instance": 2,
      "name": "Lynx Smart BMS Top",
      "power": 0,
      "soc": 98.37999725341797,
      "state": 0,
      "voltage": 13.510000228881836
    },
    {
      "active_battery_service": true,
      "current": 0,
      "id": "com.victronenergy.battery.lynxparallel0",
      "instance": 4,
      "name": "Lynx Smart BMS Totals",
      "power": 0,
      "soc": 99.18499755859375,
      "state": 0,
      "timetogo": 864000,
      "voltage": 13.505000114440918
    }
  ]
}
```

mman, 2024-05-22:
@martinbosma What would be the easiest way to simulate such system? The fix looks easy enough but I need a way to verify it works as expected.

I'm looking at https://github.com/victronenergy/venus-docker and it does not seem to contain any pre-recorded Lynx BMS configs, or does it?

mman, 2024-07-22:
Fixed via https://github.com/victronenergy/venus-html5-app/pull/371

closed 2024-07-22 by mman