victronenergy/venus-html5-app #369: Add night mode

martinbosma, 2024-05-03:


mman, 2024-09-12:
Here is the initial investigation how to best/easiest implement the functionality:

1. The `Theme` module https://github.com/victronenergy/victron-mfd-modules/tree/master/src/Modules/Theme needs to add support for `Light`, `Dark`, and `Night` theme variants. It is currently done as a simple boolean switch `Light` / `Dark` that is used in all the previous variants of the app. To make the old code compile I'd figure out a way to provide three theme variants where `Dark` and `Night` would map to the existing `darkMode` property being `true`.
3. The main logic of switching themes lies in the `App.tsx` where we observe the `darkMode` property and add/remove appropriate CSS class provided by `tailwindcss` to the root element. That switches the theme. So we will need here to add/remove `dark` and `night` CSS classes as appropriate. https://github.com/victronenergy/venus-html5-app/blob/7371e969e6e63a666103acf02306349de73078a7/src/app/Marine2/App.tsx#L58-L64
4. All elements within the app currently use the traditional `tailwindcss` styling in the form of specifying a default color, and its dark variant. Like for example here: `text-victron-gray dark:text-victron-gray-dark`. https://github.com/victronenergy/venus-html5-app/blob/7371e969e6e63a666103acf02306349de73078a7/src/app/Marine2/components/boxes/BatteriesOverview/BatteriesOverview.tsx#L76. We can simply add a variant for `night` CSS class, but that would mean modifying around ~100 entries in 33 files. Therefore point 5.
5. We should extract all those ~100 entries into symbolic colors provided via `tailwind.config.js`, I think majority of them may already be there. And then use CSS variables to customise each of these colors for `default`, `dark`, `night` mode. Finally we can remove the `dark:` prefix in all instances and only specify a symbolic color everywhere.

mman, 2024-09-12:
Additionally, instead of switching between Light/Dark/Night modes, it would be better to make Night mode a variant of the Dark mode via toggle. That way user can still opt into Light / Dark automatic theme switching and when Dark mode kicks in, the app goes into Night mode when the switch is toggled.

mman, 2025-03-12:
Addressed by https://github.com/victronenergy/venus-html5-app/pull/473

closed 2025-03-12 by mman