victronenergy/venus-html5-app #384: connect to the MQTT broker through the nginx websocket

jhofstee, 2024-06-14:
In the near future (optional) encryption and authentication will be added for the LAN services, see https://github.com/victronenergy/venus/issues/1255. Both will be handled by the webserver (nginx). In order to make at least authentication work, the venus-html5-app should no longer connect directly to the MQTT websocket, but through the webserver instead. The url is http://[host]/websocket-mqtt or https://[host]/websocket-mqtt.

The current Venus releases already include mentioned path. Accessing the app over https:// will redirect to http:// at the moment.

mman, 2024-07-22:
Fixed via https://github.com/victronenergy/victron-mfd-modules/pull/4

closed 2024-07-22 by mman

mr-manuel, 2024-08-06:
Does this also fix the connecting loop, when opening the MFD app via https?

[image]

[image]

mman, 2024-08-06:
@mr-manuel How are you accessing the app please? What browser do you use and if it is Chrome, Firefox, or Safari, could you open a web development console and see what messages get logged?

mr-manuel, 2024-08-06:
I tried with Firefox and Edge on Windows. Tried with FQDN and IP, to see if it makes a difference, but it makes not. With HTTPS it doesn't even try to make a websocket connection. In the console there is no helpful error.

Also strange that it tries to connect to port `7890` on HTTP. Probably some old code somewhere?

## Firefox HTTP working
[image]

[image]

## Firefox HTTPS not working
[image]

[image]

## Edge HTTP working
[image]

[image]

[image]

## Edge HTTPS not working
[image]

[image]

mman, 2024-08-06:
Thanks for the screenshots. How did you set up the https please? 

From the logs you sent it looks like the app is actually correctly connecting over new MQTT web socket, so that is good… but it appears to be dropping connection at some point… which is a different problem not addressed by this bug.

the fact that edge does not trust your https certificate seems also unrelated,  but is about how is the https setup… have more for me to share?

mman, 2024-08-06:
Also, the 7890 is spurious unused port, will be fixed in the next build.,.

mr-manuel, 2024-08-06:
I installed Venus OS v3.50~13 and selected the security profile `weak`, to be able to test with HTTP and HTTPS.

mman, 2024-08-07:
@mr-manuel thanks for the steps, that is precisely what I need, will try to reproduce and see what to do about it next week, I am OOO at the moment

mman, 2024-08-13:
@mr-manuel So if my understanding/testing is correct, then accessing the HTML5 app via HTTPS is not going to work at all. It's because of https://github.com/victronenergy/venus-html5-app/issues/392 where accessing insecure `ws` content when connected via `https` is refused by the browser.

I have a fix ready that allows accessing the `/app` via https properly, but it most probably does not address your connection loop issue. Could you please open another bug against this repository and add more details of what exactly are you doing and when the first connection loop occurs, and (if applicable) where the loop does not occur? I will investigate.

mr-manuel, 2024-08-13:
The connection loop is related to the same problem, so no new issue needed. Saw your opened issues. Thanks!

mman, 2024-08-15:
@mr-manuel Could you tell me more details about when you first seen the connection loop problem, and if you can reproduce it reliably? And then what should I do to reproduce it?

The thing is that currently released beta of Venus 3.50-13 only supports `http://` access to the app, and once you start using `https://` it will do nothing - it will spin with "Connecting" but the web browser log will be empty. The screenshots and logs you sent show entirely different behaviour. I'd love to learn more about how you trigger it.

Could you please send me more details?
Martin

mr-manuel, 2024-08-15:
Connection loop was probably the wrong wording. The MFD app is not connecting at all with https. I wrote connection loop because the app shows connecting forever.

mman, 2024-08-15:
> Connection loop was probably the wrong wording. The MFD app is not connecting at all with https. I wrote connection loop because the app shows connecting forever.

Yeah, that I understand... but the screenshots you have attached to this bug show different behaviour from what is expected in 3.50-13, so I wonder if there is anything else you did and as I do not want to speculate, I'd like you to explain to me step by step what steps did you do and what did and did not happen... and preferably attach a web browser development console with messages...

If all you did is updated to 3.50-13, enabled *Security Profile* *Weak* or *Secure* and tried to access the `/app` via `https://` then this is not supposed to work.

mr-manuel, 2024-08-15:
> If all you did is updated to 3.50-13, enabled Security Profile Weak or Secure and tried to access the /app via https:// then this is not supposed to work.

Yes, this are the only steps I did. So the question now to me, should it work out of the box or not?

mman, 2024-08-15:
> > If all you did is updated to 3.50-13, enabled Security Profile Weak or Secure and tried to access the /app via https:// then this is not supposed to work.
> 
> Yes, this are the only steps I did. So the question now to me is, should it work out of the box or not?

Nope, this currently is not supposed to work. Will be fixed in later beta build of Venus 3.50