victronenergy/venus-html5-app #389: Change how Remote Console is handled: QR code fallback

mpvader, 2024-07-11:
## Intro:

Some (older) MFDs can not show our webassembly version / gui-v2. Details of that are here: https://github.com/victronenergy/gui-v2/issues/909.

The tech used for showing gui-v1 is VNC. That is available for Qt4 (gui-v1), and no longer available (without going for a commercial Qt license - which is something we're not going to do) for Qt6.
  
At some point, in half a year orso, when we're moving from OE Dunfell to a newer version, we'll hopefully drop Qt4.

Therefore we need another solution for those (older) MFDs.

## Work to be done:

- [ ] Decide if we need this for gui-v2 release, ie. Venus OS v3.50, or only later, when we move away from Dunfell (@jhofstee / @mpvader )
- [ ] When only later, then keep gui-v1 on Remote Console running somewhere and use that url here in html5-app. (@jhofstee / @martinbosma )
- [ ] See if it is possible to implement a fallback option, ie. for those MFDs that do support webassembly, show gui-v2. And for those that don't, show a QR code with a URL that opens a webbrowser with gui-v2; loaded from URL for GX device. (@martinbosma / @mman)

mpvader, 2024-07-11:
@jhofstee pls review above + give your input on the first work item. If we can do that only later then that would be nice.

It does mean gui-v1 needs to remain running (is there a concern wrt CPU usage?) on all devices.

mpvader, 2024-07-11:
I've discussed above with Jeroen; and its prefered to make the cut now, ie. from html5 app remove the dependency of remote console.

mpvader, 2024-07-11:
wrt the fallback and detection: perhaps javascript can see if the browser has webassembly support; something for @mman  ?

martinbosma, 2024-07-12:
Perhaps something like this could be used: https://stackoverflow.com/questions/47879864/how-can-i-check-if-a-browser-supports-webassembly

mman, 2024-07-22:
@martinbosma @mpvader I have put together a minimal/low overhead test to add "WebAssembly" support onto diagnostics page, from there it should be fairly easy to use it elsewhere to provide the QR code fallback.

My concern is however that QR code would be referring to a URL pointing towards Cerbo GX and the iPhone/Android scanning that QR code needs to be on the same network as Cerbo, and that needs to be explained along the QR code as well I guess...

martinbosma, 2024-07-23:
@mman The description could be as simple as: "Scan the QR code to open the Remote Console in a browser. Make sure your mobile device is on the same network as the GX device." It could be in the same manner as the other instruction page we have, which we show when MQTT is not enabled.

mman, 2024-07-24:
@martinbosma @mpvader I also pondered a bit around an alternative where a QR code could be specifying a Cerbo WiFi network to join, and if that network would run (which it does not) a Captive Portal it can automatically open the browser with console preloaded... I'm not convinced it's better than the wording @martinbosma put in the comment above... just throwing it here from my head... :)

mman, 2024-09-05:
@mpvader I have drafted rough implementation in https://github.com/victronenergy/venus-html5-app/pull/408, my ignorance of not knowing anything about typical MFD deployment scenarios plays a role in my comments.

However, assuming a mobile phone is always connected to cellular, and to a boat WiFi, in order for the QR code to work, the HTML5 app running on the MFD **must absolutely know somehow about the WiFi**.

mpvader, 2024-09-06:
The QR code can simply link to https://venus.local/app/ or http://venus.local/app; which of the two depends on how the page itself is loaded. When loaded as https, show the https link; when loaded as http, show the http link.

mpvader, 2024-09-06:
ps. its not /app, since that points to html5 app instead of gui-v2.

@jhofstee , to what url should html5 link? In other words, what url will host (or redirect) to gui-v2?

mpvader, 2024-09-10:
@mman , on dbus, here is the path that can tell the html5 app what version of the gui is running:

com.victronenergy.settings/Settings/Gui/RunningVersion

1: old gui
2: gui-v2

mpvader, 2024-09-10:
and as discussed, if above settings is, old gui, behaviour is same as today.

And if that setting is set to 2, gui-v2, show the QR code, with a text "To access the GX Remote Console, make sure your phone is on the same wifi network as the GX device and then scan this QR code."

And below the QR code, show the url: https://venus.local/  (or http://venus.local), so that people with a laptop and not a phone also understand where to go.

cc @martinbosma

martinbosma, 2024-09-10:
@mpvader If people change the host name, does this affect the url? If so, shouldn't we take care of that? Or is this a corner case we can simply ignore?

mpvader, 2024-09-10:
Which host name? Anyway, no: its venus.local; not something else.

I don’t know how to change it by the way.

mman, 2024-09-10:
@mpvader Please take a look at the pictures attached to the PR https://github.com/victronenergy/venus-html5-app/pull/408 and let me know what you / the designer(s) think about the sizing of the elements and wording. 

If satisfied with the wording, I'll push it for translations and add any necessary changes to the design.

The functionality works nicely showing this only when Gui v2 is set as default and displaying a v1 console otherwise.

So that is great actually!

mpvader, 2024-09-10:
Great indeed! Looks good to me

mpvader, 2024-09-25:
@mman can you give an update: when can we expect this to appear on the todo for inclusion in Venus OS?

Or is there something that needs checking or discussion?

mman, 2024-09-25:
@mpvader I was waiting for the translations that are already uploaded in the poeditor. If you want me to merge and release with English only let me know, otherwise the code is ready.

closed 2024-09-27 by mman