victronenergy/venus-html5-app #391: Add pending state for Lynx Smart BMS

martinbosma, 2024-07-24:
For the redundancy feature, a new state has been added to the Lynx Smart BMS: pending. This means that it is waiting to close its contactor. In order display this state, it first must be added to the list of batteries, available on `N/<vrm id>/system/0/Batteries`.

martinbosma, 2024-07-26:
@mman This has been added to [systemcalc](https://github.com/victronenergy/dbus-systemcalc-py/commit/c26994f7b366bad17a1b2a89dc521780fa57498b). You should be able to use the venus-docker to simulate this.

mman, 2024-08-15:
@martinbosma Which `Venus-docker` simulation do you suggest to use? The `lp` one we added couple months back is broken and does not property display the Lynx BMS instances...

martinbosma, 2024-08-26:
The `lp` simulation should be the one. I will have a look why it is broken and try to fix it.

mman, 2024-08-26:
Thanks @martinbosma, please ping me when you think is ready... I will re-test

closed 2024-09-03 by mman