victronenergy/venus-html5-app #394: MFD unable to connect to `/app` when Security Profile is set to `Weak` or `Secure`

mman, 2024-08-13:
When `Weak` or `Secure` Security Profile is activated (version 3.50+), Venus `nginx` is protected with a password. No matter whether the `/app` is then requested via `http://` or `https://` the first load of the app on a MFD device requires password to be entered.

This is more of a reminder bug that we may want to address in the future depending on what the need is:

[image]

mman, 2024-08-29:
Marking as will not fix for now.

mwielgoszewski, 2024-12-24:
FYI - I was able to get this working with my Garmin using the new "secured" profile by just allowlisting the IP address range of the Garmin ethernet network in the nginx config. I've got a short write up that you may consider implementing: https://gist.github.com/mwielgoszewski/aef6b729cf55fa18c96d35a2d828c49b

In my configuration, I omit using the `auth_request` module entirely. Another way to handle this would be by leveraging the `satisfy any` directive with `auth_request` and `allow <ip>`. This way, you can allowlist any devices that are directly connected to the ethernet network while requiring password auth from devices that connect to it over the wifi interface.

mman, 2025-01-08:
Good suggestions @mwielgoszewski, will see which of these will work the best out of the box without much tweaking...