victronenergy/venus-html5-app #398: Disappearing and re-appearing services are not properly picked up

martinbosma, 2024-08-28:
https://github.com/user-attachments/assets/d480c79a-2db1-4c9f-8fdf-64b18119fd39

mman, 2024-08-29:
Looking at the code there does not seem to be an obvious omission that would mean that something does not get updated. 

But, there is an interesting edge case where when multiple devices are subscribing over MQTT to the same Venus device, then one of them can populate UI too early without receiving all requested MQTT values as discussed here (https://github.com/victronenergy/victron-mfd-modules/pull/8).

Because MQTT will then publish only changed values, there may be a chance (or at least this is how I understand it) where the UI can get stuck displaying uninitialised value of 0.0A and waiting for changed value from the alternator, that may not come for a very long time assuming the alternator current stays the same. Reloading will force full re-subscribe of all MQTT topics and will therefore receive a fresh update.

This should however happen only in deployments where multiple observers would subscribe over MQTT to the same Venus. Not sure if that could be the case here.

Fixing https://github.com/victronenergy/victron-mfd-modules/pull/8 may actually fix this problem.

@mpvader Do you have more details about this concrete setup and what else apart from the HTML5 app is involved?

mman, 2024-09-19:
I believe this may have been a bug that is now fixed by: https://github.com/victronenergy/venus-html5-app/commit/8a9e5b6ff3e61951a71da18654b83eeec4b2b153

mman, 2024-10-02:
Reopening, found this is a problem in multiple components that forgot to declare they are `mobx` `observer` and thus never received updates. They only got re-rendered as a side effect of other updates.

mman, 2024-10-03:
I am confident it is fixed by https://github.com/victronenergy/venus-html5-app/commit/c37ec0861ebde6eb4ef1d8450c0d99238126d5a5

closed 2024-10-03 by mman