victronenergy/venus-html5-app #400: Sort tanks on type instead of name

martinbosma, 2024-08-28:
It needs to mimic the behavior of gui-v2

[image]

mman, 2024-08-29:
The offending code in HTML5 app lives here: 

https://github.com/victronenergy/venus-html5-app/blob/77079345381847fcdb3b2b114689d7116ff8c14e/src/app/Marine2/components/boxes/Tanks/Tanks.tsx#L25-L26

We get all DBUS instance numbers representing tanks and iterate over them in random order, or better said in the order that MQTT query `N/${portalId}/tank/+/DeviceInstance` gives us back.

This code needs to be modified to fetch tank `fluidType` and `customName` so that we can actually use them to sort in some deterministic order.

Question remains, what is the actual order that the gui-v2 uses.

mman, 2024-08-29:
@martinbosma Could you please connect me with somebody that can help me understand what order gui-v2 currently uses to sort tanks, or where the code lives?

mman, 2024-08-30:
Investigation: In the gui-v2 the order of the tanks on the *Brief* page can actually be changed via *Settings / Display & Language / Brief view levels*. The default order is *Battery*, *Fuel*, *Fresh water*, *Black water*.

[image]

On the *Levels* page however the order seems to be hardcoded based on types specified here:

https://github.com/victronenergy/gui-v2/blob/e154bec791c640dd483341b4c63579b24301e4c1/data/Tanks.qml#L12-L25

[image]

So to address this issue we need to sort all returned tanks by their type in the same order gui-v2 does.

mpvader, 2024-08-30:
The setting of the order in gui-v2 only affects the brief page. Which is really something else.

html5 app needs to mimic gui-v2 levels page

DanielMcInnes, 2024-09-02:
Hi @mman , I can confirm that you are correct, the order of tanks on the Levels page is specified as you describe:

https://github.com/victronenergy/gui-v2/blob/e154bec791c640dd483341b4c63579b24301e4c1/data/Tanks.qml#L12-L25

blammit, 2024-09-02:
@mman the order is based on the tank type, as Daniel mentioned; when you have multiple tanks of the same type, then within that group, the tanks are sorted by custom name. 

If a tank does not have a custom name, then a name is created based on the fluid type and device instance are used e.g. "Fuel tank (334)", and the sorting will be done based on this name. See code [here](https://github.com/victronenergy/gui-v2/blob/main/data/common/Tank.qml#L90). (There is an extra fallback for product name there, but it is unlikely to be used as the fluid type and device instance should be available.)

Here is an example:

[image]

mman, 2024-09-02:
Thanks @DanielMcInnes and @blammit for the confirmation, much appreciated.

closed 2024-09-13 by mman