victronenergy/venus-html5-app #401: Strange jumping of battery widgets

martinbosma, 2024-08-28:
See at 14s in below video.

https://github.com/user-attachments/assets/d6a93216-54eb-486f-b0bb-693bac3afb65

mman, 2024-08-28:
I think this one is caused by the fact that 24v Lynx Smart BMS goes from _Charging_ / 100% to _Idle_  / 100% and therefore changes position.

The algorithm to sort batteries currently works as follows:

1. active battery service goes first
2. _charging_ goes before _discharging_ and that goes before _idle_
3. if they have the same parameters (active/nonactive) (charing/discharging/idle) they are sorted by their id

So the question perhaps is that we need to revisit how to actually sort all the available battery services in the system so that it makes more sense and does not possibly jump because battery state changes ...

mman, 2024-08-29:
@mpvader Do you have an idea how you would like the sorting algorithm to work to prevent accidental jumping and still be useful? Perhaps just sorting batteries with `active_battery_service: true` first and everything alphabetically by name? or id? ignoring the `state` and `soc` which can change over time?

mman, 2024-09-03:
Addressed by https://github.com/victronenergy/venus-html5-app/commit/38883e1a04650d4e7ee9e80ac16115bfe13184d2

closed 2024-09-03 by mman