victronenergy/venus-html5-app #406: Different size battery gauge

martinbosma, 2024-09-04:
In below screenshot the first battery gauge is bigger than the others, while the second one contains the same data, except for the name,
[image]

mman, 2024-09-11:
@martinbosma What simulation is that please and what OS/browser did you use to trigger it?

martinbosma, 2024-09-11:
It's the `bct` simulation and I'm using Firefox.
[image]

mman, 2024-09-11:
Got it, happens only after the battery widget reaches certain size, but I can reproduce it in Safari as well as in Firefox.

https://github.com/user-attachments/assets/e9a19922-0931-4020-96d2-f1068439caab

The bigger green circle is 700pt at its maximum while the smaller ones are at 600pt when they break.

[image]

Looks like the flex layout is getting confused by the labels where first one captures more space and therefore determines what would be the size of the circle. /me going to learn about the flex box thingy.

commit 8eb1f86f (2024-09-12): fix(battery): make sure all battery circles are always the same size, closes #406
closed 2024-09-12 by mman