victronenergy/venus-html5-app #419: chore: bump eslint to the latest version

mman, 2024-09-09:
also not as simple as it looks...

mman, 2024-10-04:
a bit of history. Venus HTML5 app was based on https://create-react-app.dev which started as a promising wrapper around `webpack` and its plugins but ended up in a state where it is no longer maintained and stuck on couple older dependencies.

More over when using CRA, one can not easily override the defaults so a process called `npm run eject` was performed in order to basically discard CRA and produce bunch of complicated `./config/` files and `./scripts` that perform the build and testing.

Notably the scripts in `./scripts` folder depend on couple outdated dependencies that are not updated for eslint 9 and that apparently are not being worked on.

One of them is https://github.com/facebook/create-react-app/tree/main/packages/eslint-config-react-app that hard depends on eslint 8 and is being hardcoded in the `webpack.config.js`

And there are couple more that are hardcoded in the webpack.config.js not doing much real work, just complicating the update process.

Neoxs, 2025-03-22:
🚀 Just completed an ESLint v9 migration, and wow… what a ride! 😵‍💫

I ran into breaking changes, weird errors, and unexpected surprises—but I documented everything so you don’t have to suffer the same way.

If you’re upgrading (or still hesitating), this guide might save you some headaches:
📖 [neoxs.me/blog/migration-to-eslint-v9](https://www.neoxs.me/blog/migration-to-eslint-v9)