victronenergy/venus-html5-app #433: Update help text when mqtt is disabled

martinbosma, 2024-09-25:
Part of https://github.com/victronenergy/venus/issues/1255

mman, 2024-09-27:
Code ready, updated the screenshot, @martinbosma Changed one string that now needs translation update. Let me know if you want to merge it with English only or whether you can arrange the translations to be updated fast.

I mage the GUI version to default to v2 so whenever MQTT is not available the `Remote Console` button now goes to display the QR code, which is the behaviour we want for the future.

[image]

martinbosma, 2024-09-27:
@mman Fine to merge it an make a release asap to also have some field testing time. Not for this, but for the other changes. If the translations are done before v3.50 is released, we can make a new release including those.

mpvader, 2024-09-30:
Sorry, another change is needed:

This error needs to be replaced with:

Title: Could not connect to the GX device
Text: Error code ... ?   (@martinbosma you can pick an error code for that)

---

background: this error will most likely never be shown any more to a user. So, hence above simple solution and what we'll then do is mention this in the GX device manual.

mpvader, 2024-09-30:
ps. per v3.50 there is another common place where users might end up: trying to access http://[IP]/app while security profile is set to secured.

We'll fix that in Venus OS itself, by telling user what to do, see here: https://github.com/victronenergy/venus/issues/1351

martinbosma, 2024-09-30:
So, we drop the screenshot? And from what "pool" of error codes should we choose? I believe there are VenusOS error codes, e.g. flash corrupted. That one, or?

Reopened #433<https://github.com/victronenergy/venus-html5-app/issues/433>.

mpvader, 2024-09-30:
> So, we drop the screenshot?

yes 

> And from what "pool" of error codes should we choose? I believe there are VenusOS error codes, e.g. flash corrupted. That one, or?

yes, that one

@martinbosma

mman, 2024-09-30:
@martinbosma @mpvader This error screen appears when there is no TCP (WebSocket) connection to the Cerbo GX..., actually no other error can cause this currently.

`Unable to connect to GX Device` sounds like a good title...

and for an explanation at the moment the only thing how to trigger the error is to disable MQTT Access... so a word about having MQTT Access enabled seems vital to me...

The app is actually loaded from the Venus GX device so in order to display this screen the MFD actually has a working http connection to Cerbo GX... what is not working is the MQTT WebSocket only and that can only be disabled by disabling MQTT Access...

thoughts?

mpvader, 2024-09-30:
As discussed with Martin M just now:

The setting in the screenshot, called 'MQTT Access', is not related to this.
In this new version, its super rare for a user to get this. So lets just make it very simple but do add an error code.

mman, 2024-09-30:
@martinbosma Let me know what `error code` /  `error text` we want to include on this screen and I will integrate it quickly...

martinbosma, 2024-10-01:
@mman How about:
Title: `Could not connect to the GX device`
Text: `GX Error #60.`

Perhaps we can refer to the [troubleshooting section](https://www.victronenergy.com/media/pg/Cerbo_GX/en/troubleshooting.html#UUID-03e33c9c-3dc3-93f7-6123-af9a567566e3) of the GX manual (not linking it) as the error code needs to be documented there.

mman, 2024-10-01:
This is the screenshot. Remote Console now shows the QR code by default, restart the app will attempt to reload everything and re-connect.

[image]

Sounds good?

mman, 2024-10-01:
@martinbosma @mpvader Screenshot above, strings updated and pushed to poeditor.

mpvader, 2024-10-23:
this has been done

closed 2024-10-23 by mpvader